ArduPilot/ardupilot #6236: QuadPlane Altitude Loss After Transition in AUTO Mode

ferdicakmak, 2017-05-13:
#### Issue details
Hello everyone,

I have performed over 30 flights with my quadplane setup using Arduplane V3.7.1 and didnot have an altitude problem. When I installed Arduplane V3.8.0beta5 I experienced altitude loss problem at 4th  flight, where previous 3 flights were ok.

4th flight: I have started an auto mission as in previous 2 flights. Quadplane took off autonomously to 20m in VTOL mode and started transiton to WP2 at 50m. After transiton is complete and quadplane fliying in FW mode, GCS reported "Reset alt target to 1610.4" message, and aircraft started to loose altitude. I have recovered from crash by changing the mode to QHOVER and landed the aircraft manually.

[image]

I dont have flash logs due to bad logging, but telemetry logs are provided below. One thing I noticed while examining .tlog is that "alt_error" becomes very big from "nav_controller_output" as shown in the graph below.

[image]

Previous 3 flights are ok considering "alt_error" values, as shown below.  

[image]

I believe that the navigation solution might have malfunctioned by reporting "Reset alt target to 1610.4" message, although it should be around 1047m.

Any advice is welcome.
Best Regards,
Ferit.

#### Version
ArduPlane v3.8.0beta5

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
QuadPlane

#### Hardware type
Pixhawk

#### Logs
[22_2017-05-12 10-11-24_1_2_3th_flight.zip](https://github.com/ArduPilot/ardupilot/files/998612/22_2017-05-12.10-11-24_1_2_3th_flight.zip)

tridge, 2017-05-15:
@ferdicakmak do you have the dataflash log from the microSD ?

ferdicakmak, 2017-05-16:
Unfortunately, I dont have dataflash logs due to "No IO Thread Heartbeat" and "Bad Logging" problem. This issue is discussed in [https://github.com/ArduPilot/ardupilot/issues/5597](url).

OXINARF, 2017-05-16:
@ferdicakmak Just to clarify, you don't have logs because you have some IO issue, probably with your SD card. The "No IO Thread" and "Bad Logging" info are warning you of that, they aren't an issue.

Without a dataflash log this is probably going to be hard to diagnose (and by the way, you should always post in the forum first, this issues list is for confirmed bugs and feature requests).
@tridge?

closed 2021-07-25 by IamPete1