ArduPilot/ardupilot #6237: Feature Request: Add "Distance to HOME", SNR, and RemSNR in Log/RAD

dongfang-a, 2017-05-13:
#### Issue details
I'm trying to measure my 3DR Radio Telemetry max distance, If there is a "Distance to HOME", SNR(Signal-to-noise ratio, RSSI - Noise), and RemSNR(Remote Signal-to-noise ratio, Remote RSSI - Remote Noise) in the Log file, I should be able to measure that easily by drawing these 3 data.

#### Version
3.4.6

#### Platform
[ ] All
[  ] AntennaTracker
[*] Copter
[  ] Plane
[  ] Rover

#### Airframe type
N/A

#### Hardware type
pixhawk

#### Logs
N/A

OXINARF, 2017-05-15:
We log all of those in RAD packet, except the distance to home. You can calculate that distance though as home and current position are both logged.

Mission Planner should also help you with this, you can activate in the HUD the distance to home as well as the radio stats. It also has a estimation of the distance left based on radio RSSI.

Pedals2Paddles, 2017-05-15:
Having DTH in the log would be quite useful. It would save having to export and calculate it, which is a lot of steps. I think it should be kept in mind, maybe the next time someone is working on a related matter?

OXINARF, 2017-05-17:
I've asked in Gitter what people thought of logging the distance to home and the general consensus was that we shouldn't do it as it can already be calculated.

Since others are already logged I'm going to close this.

closed 2017-05-17 by OXINARF