ArduPilot/ardupilot #6238: Auto-Take-Off Vertical climb then loop and crash.

Elios007, 2017-05-13:
#### Issue details
HI, 

We did more than 100 flights with auto take off with a bungee without a single problem, using the pixhawk 1 and the 3.7.0. 
Lately we have upgraded 3.7.1, we did the first fight without a problem, file 103.BIN. The landing was a bit tough with the parachute due to strong wind, we lost the stopper of the folding propeller.  We tightened the propeller very well and we tried to do the auto-take off with the bungee, but the plane went very stiff vertical and then it tried to stabilize itself but then we opened the parachute but it was too late and it crashed. file 106.bin

We did a new plane with a new AP and an APC propeller, we did all calibration and we checked all the surfaces before flights and everything was OK, but when we did the auto-take off, the plane behaved the same with vertical take off and tried to do a loop and then a crash. 24.bin 

We need please to know what is the issue, because all our flights and take off with bungee are the same for the last 3 years. 

#### Version
3.7.1

#### Platform
Plane V-Tail

#### Airframe type
Fixed Wings V-Tail Talon 

#### Hardware type
Pixhawk 1 with arduplane 3.7.1

#### Logs
24.BIN https://www.dropbox.com/sh/ilsddb3sblkmiay/AADUzUe-PCc0RcvHwg0-vHXNa?dl=0

106.BIN https://www.dropbox.com/s/ybwqqsjsah5h47u/106.BIN?dl=0

103. BIN https://www.dropbox.com/s/mq27vez3yt2bgf8/103.BIN?dl=0

magicrub, 2017-05-13:
do you have the same problem if you go back to v3.7.0?

Elios007, 2017-05-13:
We didn't try to do it, because the first flight 103.BIN was a good flight with the 3.7.1 without a single problem. 

We thought the issue is not related to the 3.7.1. But we can try it and see but we don't want to lose a plane again. 

Are you able to see any issue with the today flight 24.BIN logs ?

magicrub, 2017-05-13:
github is for confirmed bugs or feature requests. If you have a problem you'd like to get someone else to debug the best place is discuss.ardupilot.org. Please CC me on your post and I'll take a look there.

closed 2017-05-13 by magicrub

Elios007, 2017-05-13:
http://discuss.ardupilot.org/t/auto-take-off-vertical-climb-then-loop-and-crash/17437