ArduPilot/ardupilot #6243: Copter: FLOW_ENABLE should remain "1" even if we fail to connect to flow sensor

rmackay9, 2017-05-15:
If we fail to connect to the flow sensor, we set the FLOW_ENABLE parameter to zero.  The problem with this is that it leads to us silently failing when really we should leave the sensor driver active and enabled but report it as unhealthy in the extended system status mavlink message to the GCS.

This is the problem lines are these two:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_OpticalFlow/OpticalFlow.cpp#L108
https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/sensors.cpp#L292

OXINARF, 2017-05-15:
I agree, we shouldn't disable the library just because the sensor wasn't found. We should carefully review if there is code relying on this though.

bys1123, 2019-05-12:
Is FLOW_ENABLE removed on 3.7.0-dev?

rmackay9, 2019-05-12:
@bys1123, yes, it's renamed to FLOW_TYPE.

rmackay9, 2019-05-12:
Looks like we've resolved this problem so so closing!

closed 2019-05-12 by rmackay9

bys1123, 2019-05-13:
This information for others who need config FLOW_TYPE 
```
OpticalFlowType {
        NONE = 0,
        PX4FLOW = 1,
        PIXART = 2,
        BEBOP = 3,
        CXOF = 4,
        MAVLINK = 5,
        SITL = 10
}
```