ArduPilot/ardupilot #6244: RFE: Allow for airspeed recalibration with throttle/rudder joystick

marcmerlin, 2017-05-15:
As discussed here http://discuss.ardupilot.org/t/feature-to-auto-reset-airspeed-to-0-when-armed-or-should-arspd-skip-cal-disabled-do-this/17354/5 @magicrub suggested that we could add a joystick movement to allow re-setting the airspeed calibration so that airspeed reads 0.
This is supposed to be done at boot as long as ARSPD_SKIP_CAL is left to 0 (which is my setting), but I'm still getting occasional airspeeds in the 10km or more before takeoff with the pitot tube blocked, especially on one aircraft with a pixfalcon and a 3rd party airspeed sensor since 3DR wasn't selling any anymore when I needed one.
If I have apmplanner, there is a preflight airspeed reset option that works, but none of the android flight planners support this, and Tom suggested we could use the joystick to achieve this.
Since down and right is arm, how about up and right to reset airspeed to 0?

magicrub, 2017-05-15:
implemented here: https://github.com/ArduPilot/ardupilot/pull/6250

OXINARF, 2017-05-15:
I'll just say that in Copter up and right initiates compass calibration - that's not supported in Plane and airspeed sensors aren't supported in Copter, but those can probably change in the future.

marcmerlin, 2017-05-15:
Thanks @OXINARF . Given that, what do you think @magicrub , should we use up and left instead or are you happy with it being the same given that it's a calibration for each airframe and I've never seen the need to recalibrate the compass on the airfield with my transmitter?
Also, great job for the quick patch to add this! :)

WickedShell, 2017-05-15:
I'm not sure if you guys looked at the PR, but the actual comments in the PR are for up and left... My concern is up on throttle on plane isn't fundamentally safe the way it is on copter.

marcmerlin, 2017-05-15:
@WickedShell I missed the left part.
Throttle up is not safe if you're armed obviously, but this is also meant to be used when you're disarmed, you wouldn't want this to kick in if the airframe is armed anyway.
Now if your point is that someone could forget that they are armed and start the motor unwillingly with this combo, then yes they could, but I don't personally have a big issue with this: this action is being added to do a field calibration for someone who hopefully knows what they're doing and if not, they'll learn quickly ;)

WickedShell, 2017-05-15:
@marcmerlin my actual concern is with overrides active armed/disarmed doesn't have to enter into it at all. For example with my plane on the ground manual is also override and you have full throttle control regardless of armed/safety button state. Now almost no one uses overrides so this might be fine, but it's a risk. Not opposing it because so few people use the system like this, but it is a risk. (I'd prefer if we could use some other stick motion that doesn't require pushing throttle like that basically)

magicrub, 2017-05-15:
throttle-up was intentional so you *obviously* don't do it inadvertently on the ground. 

@WickedShell I don't think enabling this will conflict with your corner-case where MANUAL passthrough allows the motor to spin up. You just can't use this feature in that scenario. I don't think that will surprise anyone. If you've configured your aircraft like that then all sorts of caution signs are needed like hand-launching and such.

I knew the stick positions would be a hot topic to discuss so you'll notice I immediately added the DevCallTopic label so it will be discussed by the team. @marcmerlin you are of course welcome to attend along with everyone else!

WickedShell, 2017-05-15:
Yeah, as stated (at least on the other one) I can see that it will affect very few people and I'm not going to stand and oppose it for that reason :)

magicrub, 2017-05-16:
This pure RC input method of triggering the airspeed cal has been completely overruled on the dev call due to safety reasons: accidental triggering or not knowing the result if you don't have an OSD/GCS to confirm the results.

The team is in favor of creating a new "Flight Mode: Calibration" which allows many different stick inputs to trigger any of the current calibrations (baro, compass, gyro, accel, airspeed ect ect). This mode can either be mapped to your RC transmitter like any other flight mode or you could boot up into it using the INITIAL_MODE param and any flight mode change (or the special exit stick combo) would exit the mode

Closing this particular issue in favor of starting a new one that is based on the flight mode: Calibration, of which this will be one of any calibrations that can be performed safely.

closed 2017-05-16 by magicrub

marcmerlin, 2017-06-09:
@magicrub was there a new issue opened for that "Flight Mode: Calibration" ?
In the meantime, opened a FR with qgroundcontrol: https://github.com/mavlink/qgroundcontrol/issues/5274
and tower:
https://github.com/DroidPlanner/Tower/issues/1862
no idea if any of the two will implement it :)

If anyone finds any android GCS that support this in the meantime, please post here, thanks.

magicrub, 2017-06-10:
>was there a new issue opened for that "Flight Mode: Calibration" 

I don't see one. Mind opening one up? I think you're on your way to getting the "created the most issues" award :) But that's a good thing!

marcmerlin, 2017-06-10:
@magicrub done: https://github.com/ArduPilot/ardupilot/issues/6414
For issues, I've actually opened close 1000 android bugs at work ;)

marcmerlin, 2017-06-10:
@magicrub, by the way, is there still activity on freenode/#ardupilot for random questions, or not so much? (yes, unrelated, but don't want to ask something totally irrelevant to this issue here)

magicrub, 2017-06-10:
I've never heard of freenode for Ardupilot, all the activity is on gitter.

WickedShell, 2017-06-10:
I idle the freenode channel, it's basically dead, and I redirect most
people to gitter

marcmerlin, 2017-06-10:
@WickedShell thanks.