ArduPilot/ardupilot #6246: 3.5 Add option for avoidance stick mixing in loiter instead of minifence

roque-canales, 2017-05-15:
Hello,

It would be great to be able to use avoidance stick mixing in loiter as for avoidance in alt hold mode.

So basical be able to choose between minifence mode and stick mixing mode for loiter mode.

Today in loiter is minifence mode and in althold is stick mining.

We can also add CH6 knob for set in flight the fence margin.

What do you think?

rmackay9, 2017-05-15:
Allowing the margin to be set from the ch6 knob is pretty easy and non-controversial.

I'm not so sure about stick mixing for Loiter though.  Personally I find the stick-mixing approach a bit scary to fly with.  Maybe what's really wanted is for the vehicle to backaway from obstacles?  This is already on the to-do list.

rmackay9, 2017-05-15:
P.S. thanks for the "stick mixing" description by the way.  We didn't have a good name for the method we're using in AltHold.. "stick mixing" is a good description of what it is.. txs!

roque-canales, 2017-05-15:
Randy, also could you tell me if I set AVOID_ENABLE to 2 and if I set CHx_OPT option to avoidance, does avoidance is enable when CHx_OPT switch it's low? or it's activated? what does this CH_OPT ?

I'm ok that stick mixing is scary to use in most of case, but in certain case it be very usefull in loiter mode for expert users. Also I think that is quite simple to add this capablity ?
About the case, it's for example, in indoors with opt flow , or even with GPS in outdoors if we have bad positionning estimations.... Stick mixing could be strongger than minifence for protect drone?

About knob tune for fence margin, we need to change accordingly AVOID_DIST_MAX and AVOID_MARGIN.

iskess, 2017-05-15:
As a native Plane user I'm accustomed to Stick Mixing when momentary adjustments are needed during autoflight (especially autolanding). I recently had to fly a complex mapping mission in Copter, below the height of several smoke stacks. I did my best to flight plan around the stacks, but due to map errors I came much too close at some points.  I instinctively moved my sticks to manuver to safety, but of course that didn't do anything in Copter. When I had to resort to switching to Loiter mode to steer clear, upon reactivating Auto mode, the Copter will not return to the originally planned track. Instead it will proceed directly to the WP which messes up my flight line sidelap.  
Just a little story to explain how Stick Mixing is a useful functionality that is missing from Copter. 
As Randy points out, what I really needed was obstacle avoidance, but that still seems too complex to implement at this point.

roque-canales, 2017-05-22:
@OXINARF and @rmackay9 it will be great to have this feature on 3.5 release, could you push PR ?