ArduPilot/ardupilot #6248: 2 stage camera shutter.

geofrancis, 2017-05-15:
with arducopter you have only have one position for the camera shutter to take a photo, but on my seagullREC camera controller it has 2 modes, one channel can be used for manual photo so 1500us is idle, 1700us is focus and 1900 is shutter, in that mode you can focus once then snap the shutter much faster than with an auto focus every time.

the another channel is 1300 for video 1500 idle and 1800 for auto focus photo.

the problem is that when i try and get pixhawk to use manual photo it wont take a photo becuse it misses the focus step.

I have it working with the video/autofocus photo input but that ties up that input to only work with the shutter from pixhawk so i dont have any video control. i ended up installing a toggle switch to swap the input between the pixhawk and a spare channel on my receiver but its a hack for what is essentially a software issue.

2 solutions i can think of are 

2 stage shutter so it could focus for a set amout of time before taking a photo and lock the focus so it can shoot much faster.

shutter passthrough so the shutter channel is in RC passthrough mode unless its in auto or guided. that way pixhawk only has control when it needs it.

sashan13, 2017-09-06:
This is still very needed.

OXINARF, 2017-09-07:
@sashan13 Contributions are still very welcome 🙂

sashan13, 2017-09-07:
Okay, I can try it :)

drtrigon, 2021-06-13:
The same here have a cam that has off/picture/video.

Currently there is no way to use this with automatic trigger!

geofrancis, 2021-12-15:
can be done with scripts

closed 2021-12-15 by geofrancis