ArduPilot/ardupilot #6254: TECS: temporarily reduce roll limit during and shortly after a stall

magicrub, 2017-05-16:
During a stall the last thing we want is to perform a turn. During and shortly after a stall we should try and level off for a second or two to ensure we have control before we allow any hard turns.

An example is if a user flips to RTL during a stall in an effort to save the aircraft. Sounds like a good idea but it will likely trigger a hard u-turn at the same time which may cause it to immediately stall again.

kd0aij, 2017-05-16:
I think you want the ailerons neutral during a stall recovery, regardless of bank angle. But rudder should be used to minimize yaw rate until AOA is reduced and airspeed is regained. Once airspeed is adequate, then it's time to level the wings.

marcmerlin, 2017-05-16:
thanks for filing this. Technically @kd0aij is indeed right.
In my case, the AP started a turn after it was doing stall recovery, which it indeed shouldn't have.
That said, on most airframes, you can still put minimal aileron to recover a level altitude while in a stall. If you stall at a pitch angle of 90 degrees and the stall code takes a while to decide that the stall is over and things are good, you don't want to fall at 90 degree pitch and only rely on the rudder, which could be small and not very effective to level the plane back.
1) do not initiate turns with the ailerons during stall recovery (the problem I reported)
2) limit aileron deflection to x% of normal to try and level the plane back up while it's stalled (to take @kd0aij 's comment into account)
?

EShamaev, 2017-05-16:
By the way, I am working on AOA based prevention of stall. This calculated parameter showed very well.
@marcmerlin When it did start to turn, did you enter a secondary stall or "Turning Stall"? Is it possible to see logs?

WickedShell, 2017-05-16:
I'm sure most people here are tracking this, but ArduPlane doesn't detect a stall condition, or have recovery techniques. Thats honestly the first step towards doing recovery. So while I agree turning during recovery is incorrect the first step will be actually detecting and starting any form of recovery procedure.

The goal isn't to be a damper, I just want to make sure people are tracking there is a lot of other work to be done on stalls as well.

EShamaev, 2017-05-16:
Calculated AOA can already be used for preliminary stall detection.

marcmerlin, 2017-05-16:
@EShamaev see https://github.com/ArduPilot/ardupilot/issues/6212 and http://marc.merlins.org/tmp/bix_near_takeoffcrash.bin
Specifically see https://github.com/ArduPilot/ardupilot/issues/6212#issuecomment-301653776
The problem was that some integrator and spurious mode switching due to a controller problem apparently (If I understood @magicrub 's explanation properly) prevented AP from pitching up past -11 degrees for almost 4 seconds while airspeed was getting faster and faster and the ground ever closer :)

Stall recovery was not an issue, airspeed was recovered before the AP had time to even do anything, but it took very long to get the nose up to a positive attitude while doing a hard banked turn with the nose down (hard banked turn meaning loss of lift in a situation where the plane was already losing altitude fast) . This issue was to address my comment that AP should avoid doing turns, especially hard banked turns, while in an altitude recovery situation, even if there is no stall.

marcmerlin, 2017-05-16:
@WickedShell you are correct that this is not related to a stall, I've changed the issue title according to remove confusion. Does it look better now?

magicrub, 2017-05-16:
@marcmerlin that pitch "bug" was fixed 6 months ago but it wasn't limiting the pitch, your aircraft was clearly pitching up, it just wasn't pushing hard enough initially. Would you mind taking a look at v3.8beta? It's been in beta a very long time, sorry for the late release. We've fixed many things, I think including your ESC bug too.

marcmerlin, 2017-05-16:
@magicrub I don't want to cross bugs, but I'm not sure what you mean "clearly pitching up" it pitched up from -40-ish to -10-ish and then stayed at -10 pitch down for 3-4 seconds. That doesn't feel like "not pushing hard enough" :)
That said this RFE is indeed about avoiding bank angles in situations where the aircraft needs to recover  altitude, 
Let me try 3.8 beta now, thanks the heads up on it.

magicrub, 2017-05-16:
a given deflection should result in a pitch rate. If we see a desired pitch deflection but not a pitch change, then you have an airfoil/lift problem. Anyway, lets take this discussion elsewhere so we don't derail this issue