ArduPilot/ardupilot #6255: Plane: Why reset Pitch PID.i on every mode change?

magicrub, 2017-05-16:
Why are we resetting Pitch PID.i on every mode change. It makes sense, a little, when switching to MANUAL but if going to AUTO to CRUISE to RTL why are we having to relearn it each time?

EShamaev, 2017-05-16:
It should have been fixed by @tridge in December last year:
http://discuss.ardupilot.org/t/reset-of-integral-part-of-pids-with-mode-change/13487

WickedShell, 2017-05-16:
It was, we looked at the wrong code: https://github.com/ArduPilot/ardupilot/commit/3dcf4cfb55d93e298d2a02732d7eeeb72739f4a7

Anyways, I still think resetting integrators is possibly the right thing when exiting/entering manual (but I didn't do the orginal reset code). Separate issue though :)

closed 2017-05-16 by WickedShell

magicrub, 2017-05-16:
yippy!