ArduPilot/ardupilot #6257: what is changing

colorfuljune, 2017-05-16:
these had changed on May 15 ,  what is changing on github?

http://firmware.ap.ardupilot.org/Copter/beta/PX4-heli/
	ArduCopter-v1.px4	May 15 12:28	792K
	ArduCopter-v2.px4	May 15 12:28	814K
	ArduCopter-v3.px4	May 15 12:28	839K
	ArduCopter-v4.px4	May 15 12:28	817K
	firmware-version.txt	May 15 12:28	88
	git-version.txt	May 15 12:28	200

Platform

[  ] Copter

khancyr, 2017-05-16:
This issues list is for feature requests and confirmed bugs. For user support please post in the forum (http://discuss.ardupilot.org).
About, your question, just follow the git history or look on release tab https://github.com/ArduPilot/ardupilot/commit/376b9e994f8afa4cddca29dade8c867d57c3e948

rmackay9, 2017-05-16:
We report new beta release on the discourse server.  Here's the notification including the release notes.
http://discuss.ardupilot.org/t/copter-3-5-0-rc5-released-for-beta-testing

closed 2017-05-16 by rmackay9