ArduPilot/ardupilot #6259: Unstable Attitude Control in new firmware

SwiftGust, 2017-05-16:
#### Issue details
_Please describe the problem, or desired feature_
[image]

This is SITL-Gazebo data with coaxcopter frame.
Attitude Instablity caused in stabilize mode(with ANGLE_MAX 1000 centi-degree).
I uploaded roll data only but, all 3 axis goes off by attitude controller.
I expected attitude command always near at 0, but it caused instablity of airframe.
and caused crash later.
and due to the instability it's impossible to use automatic mode. (guided take-off, wp navigation)

I couldn't find what caused this yet.

#### Version
_What version was the issue encountered with_
3,5 rc-6 downloaded earlier today.
#### Platform
[  ] All
[  ] AntennaTracker
[V ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
CoaxCopter
#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
SITL.
#### Logs
_Please provide a link to any relevant logs that show the issue_
Forgot to take it from my lab computer (will upload later.)

SwiftGust, 2017-05-17:
Especially, Heading(Yaw) command is subject to diverge.
Confirmed through SITL, Real Frame(with fixation)

SwiftGust, 2017-05-20:
summarize it, heading controller 
regardless of flight mode
(tested it on althold, poshold, automode
particulary in automode, copter was able to do waypoints navigation and auto-land quite clean.
but, its heading was constantly rotated.
 - So I suspect yaw controller or yaw rate controller.

But in other modes, constant heading changes causes copter to crash.
So It is very important to find cause

P.S. I'm not sure this problem only happens to me or
is there anyone have experienced same

SwiftGust, 2017-05-24:
I found more problem. When throttle is less than hover.
attitude controller causes flip or oscillation of roll/pitch or heading divergence.

I think this part causes this problem 
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_AttitudeControl/AC_AttitudeControl.cpp#L429

rmackay9, 2017-05-24:
@SwiftGust,
The attitude controllers are very well tested on all the other multicopters so I think it's much more likely that the issue is in one of these places:

- motors library for Coax Copter
- tuning
- timing issues on the PC running the simulator
- issues with the simulator

Thanks very much for giving the coaxcopter a try, it's not a commonly used frame I think.

SwiftGust, 2017-05-24:
@rmackay9 
I checked CoaxCopter part. and others also I'm trying to figure it out still.
I think mixing part can be excluded since
 - Weird attitude command is logged and states are following such command.
-  And pwm out looks quite along with these roll, pitch, yaw command.
- Tuning can partially contribute to the problem. but there's also some unexplainable issue.
(such as command can be made like inverted flight with stabilize, poshold, auto... 
e.g. sudden inverted flight of roll 170 degree or exponentially increasing yaw rate)

Gazebo simulator and SITL environment are running on the same ubuntu desktop.
and there are some people reporting these kinds of issue since 3.5-rc 5.

SwiftGust, 2017-05-26:
After the @rmackay9's feedback.
I look throrugh randy's point out. and trying it on new rc7 firmware.
as far as I found out. 
It is both partially contritbutes to the problem.
1. relax_attitude_controllers() of AC_AttitudeControl. 
I felt like current status becoming desired attitude. this function sets that way in the various flight mode
It looks like doesn't make problem to common copters.

2. actuator scaling of roll & pitch actuator.
I think this kind of scaling required since roll, pitch control power scales with throttle.
but, requires some fix more reasonable with real aircraft. I guess minimum 0.5 for now.

3. But, heading problem couldn't figure it out yet.

Thanks.

lthall, 2017-05-29:
Hi Swift,
I think this is a tuning problem. It looks like the yaw controller is oscillating. I would suggest dropping the Yaw gains and turning on the PID logging.

SwiftGust, 2017-05-30:
@lthall 
I thought it was attitude control. 
since it shows desired attitude goes weird and state is simply following.
I should tune attitude controller more.
Thank you for comment.

P.S. when I was testing this I dropped every PID Gains low as possible except Rate D gains to be high.
and disabled feedforward control.

SwiftGust, 2017-05-30:
Weird heading of coaxcopter derived from different reasons,

- Throttle change certainly makes weird heading change even in stabilize mode.
actuator gain scheduler with heading is also little bit weird.
- Weird yaw behavior is also coming from get_auto_heading() of control_auto, WP_nav controller.
I made code to print get_auto_heading() when it is used in simple waypoint navigation.
it's not much weird but it actually derives little bit weird heading
it's setting to ahrs.yaw in the AP_WPNav::get_yaw()
i.e Vehicle is halfway to next WP toward south, it commands to head east.

SwiftGust, 2017-06-02:
Weird heading control is solved as far as I test it out.
Problem was also in the actuator gain of the coaxcopter as @rmackay9 pointed.
heading control was not able if throttle is high or low 
even I made yaw_headroom to500, and possible gain changes.

closed 2018-01-11 by lthall