ArduPilot/ardupilot #6261: Upgrading to AP 3.7 seems to cut the throttle output a split second during init and triggers an error on castle ESCs.

marcmerlin, 2017-05-16:
Initially reported here http://discuss.ardupilot.org/t/upgrading-2-planes-to-3-7-causes-the-castle-esc-to-triple-beep-esc-detected-unusual-signals-or-loss-of-signal-on-throttle-wire-unless-i-boot-in-failsafe-mode/17472
I did more testing with a servo controller and talked to a castle engineer.
The 3 beep code I started getting on 2 different planes after I upgraded them to 3.7 (one with pixhawk, one with pixfalcon), means (as quoted by castle) "what is happening is at some point after the esc is armed the flight controller is not sending a throttle signal - even if for a split second".
I've used a servo tester to verify that behaviour:
1) boot the ESC with the servo tester plugged in at any PWM value
2) ESC arms 
3) unplug servo tester
4) plug servo tester back in
5) ESC complains about loss of signal with 3 beep code that never ends until power is reset
6) resetting power brings me back to step #1 

The only way to make things work is to boot the pixhawk with loss of signal, then it boots in circle mode and sends throttle value to the ESC from the beginning without dropping the signal
Then turn the LRS radio on and start feeding data to the pixhawk, it gets out of circle and back to manu, then things work good until the FC is rebooted.
I did not have this problem with AP 3.5 or 3.6, so I'm pretty certain something in the code changed that causes a very quick drop of signal on CH3.
I also checked that booting the ESC with a PWM value as low as 800 is not a problem, so it's not a throttle failsafe or initial throttle value issue from the LRS system like I originally thought.

marcmerlin, 2017-05-16:
I'm not going to set the BUG label because I haven't proven that there is a bug in the code, but it really looks like a regression.

marcmerlin, 2017-05-16:
@magicrub I tried to flash 3.8.0beta 5 per your recommendation and now the pixfalcon won't boot.
It shows up as PX4 FMU 2.0 on the USB port and it makes this 3 tone sound without the 3 beeps at the end and sits in that state forever: http://download.ardupilot.org/downloads/wiki/pixhawk_sound_files/PX4_ReadyToUploadIOBoardFirmware.wav
I reflashed 3.7.1 and it's back to working thankfully. I guess it's a bit too beta for my board :)

proficnc, 2017-05-16:
are you loading v2, or v3 firmware? 

try loading a daily build, and make sure you pick v2 firmware.
http://firmware.ap.ardupilot.org/Plane/latest/PX4/ArduPlane-v2.px4

peterbarker, 2017-05-17:
On Tue, 16 May 2017, Marc MERLIN wrote:

> @magicrub I tried to flash 3.8.0beta 5 per your recommendation and now the
> pixfalcon won't boot.
> It shows up as PX4 FMU 2.0 on the USB port and it makes this 3 tone sound
> without the 3 beeps at the end and sits in that state forever:http://download.ardupilot.org/downloads/wiki/pixhawk_sound_files/PX4_ReadyT
> oUploadIOBoardFirmware.wav
> I reflashed 3.7.1 and it's back to working thankfully. I guess it's a bit
> too beta for my board :)

Crossed this with tridge.  He flies a PixFalcon on the eFlight 
Convergence, so it does work with master - at least on some PixFalcons :-)

marcmerlin, 2017-05-17:
@proficnc 
apmplanner2 2.0.24 seems to be unable to flash any firmware (looks like a new bug)
I then used mission planner, clicked beta, clicked airplane and let it flash.
I have no idea whether it picks v2 or v3 firmware or how to select one over the other.

billbonney, 2017-05-17:
use the latest master code as it's fixed in that versions.

marcmerlin, 2017-05-17:
@billbonney you mean I should run apmplanner2 master, not ardupilot master, correct? :)
So nothing built on http://firmware.us.ardupilot.org/Tools/APMPlanner/ will work,  and I have to get
https://github.com/ArduPilot/apm_planner and build it for linux. Correct?
Thanks, Marc

billbonney, 2017-05-17:
@marcmerlin Yes if you build latest master for  apm_planner on linux, it will work

marcmerlin, 2017-05-26:
@billbonney like I expected was going to be the case, building apmplanner is non trivial. Filed https://github.com/ArduPilot/apm_planner/issues/1045

marcmerlin, 2017-05-26:
@billbonney even with a new apmplanner, flashing does not work.
Started downloading http://firmware.eu.ardupilot.org/Plane/beta/PX4/ArduPlane-v2.px4
redirect to http://firmware.us.ardupilot.org/Plane/beta/PX4/ArduPlane-v2.px4
Finished downloading /tmp/APM Planner.w16710
then
"please unplug and plug back in the pixhawk"
then it dies with 
[20170525 23:57:59.452 INFO ] - Port found! "ttyACM0"
[20170525 23:57:59.453 ERROR] - Unable to open port: "Permission denied"
[20170525 23:57:59.453 WARN ] - QIODevice::write (QSerialPort): device not open
yet it has access but probably tries quicker than udev can fix the perms.

I had to work around it by running apmplanner2 as root. If you'd like a bug on this let me know.

However, things are still as broken as they were when I reported when I flashed with mission planner.
@proficnc @peterbarker : both apmplanner and mission planner select a flash a non working build on pixfalcon.

You told me to get  http://firmware.ap.ardupilot.org/Plane/latest/PX4/ArduPlane-v2.px4 manually and flash that, correct?
I did and I still get the first 3 notes of encounter of a 3rd kind in a loop

magicrub, 2017-05-26:
> org/Plane/beta/PX4/ArduPlane-v2.px4
> v2.px4
> ArduPlane-v2.px4 manually and flash that, correct?

latest is still 3.7.x, use
http://firmware.ap.ardupilot.org/Plane/beta/PX4/ArduPlane-v2.px4 for 3.8beta

marcmerlin, 2017-05-26:
@magicrub thanks. Sadly http://firmware.ap.ardupilot.org/Plane/beta/PX4/ArduPlane-v2.px4 still has the same problem.
My pixfalcon still gives me the 3 tone in a loop after I flash it.
The file says:
{
    "board_id": 9, 
    "magic": "PX4FWv1", 
    "description": "Firmware for the PX4FMUv2 board", 
    "ardupilot_git_hash": "708b483d770bd529eb978d2be4486463435b7f30",

magicrub, 2017-05-27:
That's very strange. I'm Mobile, can someone please check that hash and confirm it's plane v3.8beta?

peterbarker, 2017-05-27:
On Thu, 25 May 2017, Marc MERLIN wrote:

> @billbonney like I expected was going to be the case, building apmplanner is
> non trivial. Filed ArduPilot/apm_planner#1045

I create a Vagrantfile for apm planner a while ago.

I use preloads and the like to run it from the build without installing 
it.

https://github.com/ArduPilot/apm_planner/pull/1046

marcmerlin, 2017-05-28:
@magicrub @proficnc @peterbarker I just tried putting the latest beta as downloaded by mission planner , on my pixhawk, and same problem than with the pixfalcon.
I get the 3 first notes of encounter of 3rd kind, in a loop.
I had to reflash 3.7.1 to get it back to a working state.

peterbarker, 2017-05-28:
On Sun, 28 May 2017, Marc MERLIN wrote:

> @magicrub @proficnc @peterbarker I just tried putting the latest beta as
> downloaded by mission planner , on my pixhawk, and same problem than with
> the pixfalcon.
> I get the 3 first notes of encounter of 3rd kind, in a loop.
> I had to reflash 3.7.1 to get it back to a working state.

I believe that's the tone that indicates it is having trouble flashing 
the IO board.

Try holding down the safety switch as the board is booting; it attempts 
the flash in a slightly different way.

marcmerlin, 2017-05-30:
Sigh, while debugging this, I went to 3.8beta5 which gave me problems I will describe later.
I downgraded back to 3.7.1, and after I plugged the battery in, my motor spun up for about 1 second, before shutting down and then giving the lost RC pulse on throttle error that I opened this issue for.
Sadly, I don't have logs, the system was not even booted up or armed, and this only happened once, but it's kind of a big deal that the bug introduced in 3.7 can now spin up the motor, even if it only happened once to me.

marcmerlin, 2017-05-30:
@peterbarker ok, so I tried 3.8beta5 again. First, the safety switch trick worked, but I really hope this will not be required because one of my aircraft does not even have a safety switch (I use arm by stick). It was a big hassle to make flashing work.
Once I did, then I got compass not calibrated, had to recalibrate (and then again when I downgraded back to 3.7.1).
After all this, then I was never able to arm because something broke in the GPS code and my GPS kept rebooting/being redetected. Logs below.
Back to 3.7.1 the problem went away instantly.
3.7.1 logs:
[20170530 10:14:49.529 INFO ] - STATUS TEXT: 6 : "GPS 0: detected as u-blox at 115200 baud"
[20170530 10:14:49.529 DEBUG] - APM: Text Message rx'd "GPS 0: detected as u-blox at 115200 baud"
[20170530 10:14:49.530 DEBUG] - Severity: 6  text: "GPS 0: detected as u-blox at 115200 baud"
[20170530 10:14:49.547 INFO ] - STATUS TEXT: 4 : "EK2: Changed EK2_GPS_TYPE to 1"
[20170530 10:14:49.548 DEBUG] - APM: Text Message rx'd "EK2: Changed EK2_GPS_TYPE to 1"
[20170530 10:14:49.548 DEBUG] - Severity: 4  text: "EK2: Changed EK2_GPS_TYPE to 1"

3.8.5 logs:
[20170530 10:09:48.697 INFO ] - "Terrain Request: lat373104595,lon-1220646005 30"
[20170530 10:09:51.044 INFO ] - "Terrain Request: lat373104595,lon-1220646005 30"
[20170530 10:09:53.144 INFO ] - "Terrain Request: lat373104595,lon-1220646005 30"
[20170530 10:09:54.642 INFO ] - STATUS TEXT: 6 : "GPS 1: detected as NMEA at 38400 baud"
[20170530 10:09:54.643 DEBUG] - APM: Text Message rx'd "GPS 1: detected as NMEA at 38400 baud"
[20170530 10:09:54.645 DEBUG] - Severity: 6  text: "GPS 1: detected as NMEA at 38400 baud"
[20170530 10:09:55.383 INFO ] - "Terrain Request: lat373104595,lon-1220646005 30"
[20170530 10:09:57.595 INFO ] - "Terrain Request: lat373104595,lon-1220646005 30"
[20170530 10:09:58.645 INFO ] - STATUS TEXT: 6 : "GPS 1: detected as NMEA at 115200 baud"
[20170530 10:09:58.646 DEBUG] - APM: Text Message rx'd "GPS 1: detected as NMEA at 115200 baud"
[20170530 10:09:58.647 DEBUG] - Severity: 6  text: "GPS 1: detected as NMEA at 115200 baud"
[20170530 10:09:59.947 INFO ] - "Terrain Request: lat373104595,lon-1220646005 30"
[20170530 10:10:02.080 INFO ] - "Terrain Request: lat373104595,lon-1220646005 30"
[20170530 10:10:04.415 INFO ] - "Terrain Request: lat373039916,lon-1220646005 30"
[20170530 10:10:06.744 INFO ] - "Terrain Request: lat373039916,lon-1220646005 30"
[20170530 10:10:08.778 INFO ] - STATUS TEXT: 6 : "GPS 1: detected as NMEA at 38400 baud"
[20170530 10:10:08.779 DEBUG] - APM: Text Message rx'd "GPS 1: detected as NMEA at 38400 baud"
[20170530 10:10:08.779 DEBUG] - Severity: 6  text: "GPS 1: detected as NMEA at 38400 baud"
[20170530 10:10:08.934 INFO ] - "Terrain Request: lat373039916,lon-1220646005 30"
[20170530 10:10:11.290 INFO ] - "Terrain Request: lat373039916,lon-1220646005 30"
[20170530 10:10:12.962 INFO ] - STATUS TEXT: 6 : "GPS 1: detected as NMEA at 115200 baud"
[20170530 10:10:12.963 DEBUG] - APM: Text Message rx'd "GPS 1: detected as NMEA at 115200 baud"
[20170530 10:10:12.963 DEBUG] - Severity: 6  text: "GPS 1: detected as NMEA at 115200 baud"

Moral of the story
1) 3.8beta5 is clearly not ready (but it's a beta, so I guess that's ok)
2) the throttle bug likely needs to be fixed in the 3.7 branch given that it can be a safety issue.

peterbarker, 2017-06-01:
On Tue, 30 May 2017, Marc MERLIN wrote:

> @peterbarker ok, so I tried 3.8beta5 again. First, the safety switch trick worked, but I really hope this will not be required because one of my aircraft does not even have a safety switch (I use arm by stick). It was a big hassle to make flashing work.

OK, on the vehicle without a safety switch you can flash by logging into 
the nsh console and flashing it "by hand", so no need to break out the 
soldering iron.

> Once I did, then I got compass not calibrated, had to recalibrate (and then again when I downgraded back to 3.7.1).
> After all this, then I was never able to arm because something broke in the GPS code and my GPS kept rebooting/being redetected. Logs below.

What sort of GPS are we actually talking about?

Michael du Breuil is the person to look at the GPS issue.

@WickedShell @WickedShell @WickedShell (it is *three* times, right?)

> Moral of the story
>
>  1. 3.8beta5 is clearly not ready (but it's a beta, so I guess that's ok)

Ah, no.  It is generally believed that this beta is good-to-go.  You're 
seeing evidence otherwise.  @tridge

>  2. the throttle bug likely needs to be fixed in the 3.7 branch given that it can be a safety issue.

Agree!

marcmerlin, 2017-06-01:
@peterbarker you gave me a good laugh with the triple summon :) 
Are there instructions on how to flash "by hand" as you said? If it's going to be required for everyone upgrading on pixhawk and clones, if it can't be automated by the firmware, instructions on how to do it would be helpful

How do I find which GPS type is in the aircraft? (I can send a picture if needed), this didn't help:
mavlogdump.py RTL_Stall_3.BIN |grep GPS
2015-12-05 16:14:17.12: FMT {Type : 130, Length : 50, Name : GPS, Format : QBIHBcLLeeEefB, Columns : TimeUS,Status,GMS,GWk,NSats,HDop,Lat,Lng,RAlt,Alt,Spd,GCrs,VZ,U}
2015-12-05 16:14:17.12: FMT {Type : 131, Length : 50, Name : GPS2, Format : QBIHBcLLeeEefB, Columns : TimeUS,Status,GMS,GWk,NSats,HDop,Lat,Lng,RAlt,Alt,Spd,GCrs,VZ,U}
2015-12-05 16:14:17.12: FMT {Type : 149, Length : 39, Name : CAM, Format : QIHLLeeccC, Columns : TimeUS,GPSTime,GPSWeek,Lat,Lng,Alt,RelAlt,Roll,Pitch,Yaw}
2015-12-05 16:14:17.13: PARM {TimeUS : 310430917, Name : GPS_TYPE, Value : 1.0}
2015-12-05 16:14:17.13: PARM {TimeUS : 310430930, Name : GPS_TYPE2, Value : 0.0}
2015-12-05 16:14:17.13: PARM {TimeUS : 310430999, Name : GPS_NAVFILTER, Value : 8.0}
2015-12-05 16:14:17.13: PARM {TimeUS : 310431013, Name : GPS_AUTO_SWITCH, Value : 1.0}
2015-12-05 16:14:17.13: PARM {TimeUS : 310431027, Name : GPS_MIN_DGPS, Value : 100.0}
2015-12-05 16:14:17.13: PARM {TimeUS : 310431042, Name : GPS_SBAS_MODE, Value : 2.0}
2015-12-05 16:14:17.13: PARM {TimeUS : 310431068, Name : GPS_MIN_ELEV, Value : -100.0}
2015-12-05 16:14:17.13: PARM {TimeUS : 310431084, Name : GPS_INJECT_TO, Value : 127.0}
2015-12-05 16:14:17.13: PARM {TimeUS : 310431100, Name : GPS_SBP_LOGMASK, Value : -256.0}
2015-12-05 16:14:17.13: PARM {TimeUS : 310431117, Name : GPS_RAW_DATA, Value : 0.0}
2015-12-05 16:14:17.13: PARM {TimeUS : 310431134, Name : GPS_GNSS_MODE, Value : 0.0}
2015-12-05 16:14:17.34: PARM {TimeUS : 310640684, Name : AHRS_GPS_GAIN, Value : 1.0}
2015-12-05 16:14:17.34: PARM {TimeUS : 310640701, Name : AHRS_GPS_USE, Value : 1.0}
2015-12-05 16:14:17.34: PARM {TimeUS : 310641248, Name : AHRS_GPS_MINSATS, Value : 6.0}
2015-12-05 16:14:17.34: PARM {TimeUS : 310643438, Name : AFS_WP_GPS_LOSS, Value : 0.0}
2015-12-05 16:14:17.38: PARM {TimeUS : 310679916, Name : AFS_AMSL_ERR_GPS, Value : -1.0}
2015-12-05 16:14:17.38: PARM {TimeUS : 310679985, Name : AFS_MAX_GPS_LOSS, Value : 0.0}
2015-12-05 16:14:17.38: PARM {TimeUS : 310680516, Name : EKF_GPS_TYPE, Value : 0.0}

marcmerlin, 2017-06-01:
If that helps, I had a look at the nondescript GPS which came with that plane (didn't actually buy that one) and the antenna says "cirocomm gps l 00by"

marcmerlin, 2017-06-01:
@peterbarker can you point Michael du Breuil at this bug and/or redirect me to a proper bug where I can talk to him about that issue?
Also, I'll be honest, if I'm going to have to recalibrate my compass every single time I upgrade to 3.8 and downgrade back to 3.7, it's going to be old really fast :-/

WickedShell, 2017-06-01:
@marcmerlin Your 3.7 log indicates you were actually experiencing a GPS problem then as well. ```EK2: Changed EK2_GPS_TYPE to 1``` means that the GPS wasn't providing a vertical velocity to the EKF and the EKF if using your GPS as a 2D gps. Given that u-blox should always be providing a vertical velocity you appear to have been missing fields from the GPS.

The normal thing that gets blamed here tends to be bad cabling/the GPS not receiving the data transmitted from ArduPilot. However that shouldn't result in your GPS being detected as NMEA now. Unfortunately the antenna doesn't tell us a lot. If you can find out what is printed on the u-blox sticker on the back of the GPS that would be good.

marcmerlin, 2017-06-01:
@WickedShell thanks for your reply.
The GPS is glued into the frame. Are there AT commands or some remote way I can interrogate it?
Back to bad cabling, I can check it. That plane has been flying ok with 3.7 (including navigation and RTL) but it's possible that cabling just went bad.

marcmerlin, 2017-06-01:
Ok, on the plus side, I just recalibrated my 2nd plane (the big one with a real pixhawk and GPS from 3DR) and got it to arm with AP 3.8rc5
a) the ESC error at boot is gone
b) GPS comes up ok and plane is able to arm:
[MAV 001:1] GPS 1: detected as u-blox at 115200 baud 
[MAV 001:1] u-blox 1 HW: 00040007 SW: 7.03 (45970) 
[MAV 001:1] EKF2 IMU0 Origin set to GPS 
[MAV 001:1] EKF2 IMU1 Origin set to GPS 
[MAV 001:1] EKF2 IMU0 is using GPS 
[MAV 001:1] EKF2 IMU1 is using GPS 
[MAV 001:1] Throttle armed

WickedShell, 2017-06-01:
You could try this: http://ardupilot.org/plane/docs/common-ublox-gps.html#connection-option-1-mission-planner-and-pixhawk-as-passthrough (this does work over a telem radio, but works best if you use a USB to the PH) to connect u-center to allow you to identify the GPS, however if you have a cabling issue u-center will have trouble talking to the GPS as well. Do you know where you bought the GPS unit/if the manufacturer says what they put in?

That is what we expect to see on the second aircraft :)

marcmerlin, 2017-06-01:
@WickedShell, ok I got the bixler with pixfalcon to give me its GPS info as a message at boot.
(on the GPS question, actually now that I think about it, I think that GPS came with my pixfalcon kit, but I'm not 100% certain)
[MAV 001:1] u-blox 0 HW: 00080000 SW: 2.01 (75350) 

Boot with 3.7.1:
[MAV 001:1] Initialising APM 
[MAV 001:1] Airspeed sensor calibrated 
[MAV 001:1] GPS 0: detected as u-blox at 115200 baud 
[MAV 001:1] EKF2 IMU0 initial yaw alignment complete 
[MAV 001:1] EKF2 IMU1 initial yaw alignment complete 
[MAV 001:1] Init HOME 
[MAV 001:1] GPS alt: 13275 
[MAV 001:1] EKF2 IMU0 tilt alignment complete 
[MAV 001:1] EKF2 IMU1 tilt alignment complete 
[MAV 001:1] u-blox 0 HW: 00080000 SW: 2.01 (75350) 
[MAV 001:1] EKF2 IMU0 Origin Set 
[MAV 001:1] EKF2 IMU1 Origin Set 
[MAV 001:1] EKF2 IMU0 is using GPS 
[MAV 001:1] EKF2 IMU1 is using GPS 

I'll re-re-reinstall 3.8rc5 on it and see what it says again (a bit of a pain with the safety switch flashing and then having calibrate the compass again)

marcmerlin, 2017-06-01:
Back to 3.8rc, now it seems to work, not sure why. Maybe it's because when I last tried to boot it, it was only powered by the USB connection and that wasn't enough?
It still won't arm, but it may be the GPS's fault for being crappy. Not sure if AP 3.8 is more picky than 3.7 on that GPS
[MAV 001:1] Initialising APM 
[MAV 001:1] ArduPlane V3.8.0beta5 (708b483d) 
[MAV 001:1] PX4: 8d505a02 NuttX: 1a99ba58 
[MAV 001:1] PX4v2 00380039 33355114 35303335 
[MAV 001:1] Barometer calibration complete 
[MAV 001:1] Airspeed calibration started 
[MAV 001:1] Ground start complete 
[MAV 001:1] Airspeed sensor calibrated 
[MAV 001:1] GPS 1: detected as u-blox at 115200 baud 
[MAV 001:1] EKF2 IMU1 initial yaw alignment complete 
[MAV 001:1] EKF2 IMU0 initial yaw alignment complete 
[MAV 001:1] Init HOME 
[MAV 001:1] EKF2 IMU0 tilt alignment complete 
[MAV 001:1] EKF2 IMU1 tilt alignment complete 
[MAV 001:1] EKF2 IMU0 Origin set to GPS 
[MAV 001:1] EKF2 IMU1 Origin set to GPS 
[MAV 001:1] u-blox 1 HW: 00080000 SW: 2.01 (75350) 
[MAV 001:1] EKF2 IMU0 is using GPS 
[MAV 001:1] EKF2 IMU1 is using GPS 
(...)
[MAV 001:1] PreArm: GPS speed error 2.5 (needs 1.5) 
[MAV 001:1] PreArm: GPS speed error 2.1 (needs 1.5) 
[MAV 001:1] EKF2 IMU1 tilt alignment complete 
[MAV 001:1] EKF2 IMU0 tilt alignment complete 
[MAV 001:1] u-blox 1 HW: 00080000 SW: 2.01 (75350) 
[MAV 001:1] PreArm: Gyros not calibrated 
[MAV 001:1] FAILURE: Failed CMD: 400 
[MAV 001:1] PreArm: Gyros not calibrated 
[MAV 001:1] FAILURE: Failed CMD: 400 
[MAV 001:1] Initialising APM 
[MAV 001:1] SUCCESS: Executed CMD: 241 
[MAV 001:1] PreArm: GPS HDOP 2.7 (needs 2.5) 
[MAV 001:1] FAILURE: Failed CMD: 400 
[MAV 001:1] PreArm: GPS HDOP 2.8 (needs 2.5) 
[MAV 001:1] PreArm: GPS vert error 12.2m (needs 
[MAV 001:1] FAILURE: Failed CMD: 400 
[MAV 001:1] PreArm: GPS speed error 2.2 (needs 1.5) 
[MAV 001:1] FAILURE: Failed CMD: 400 
[MAV 001:1] PreArm: GPS speed error 2.3 (needs 1.5) 
[MAV 001:1] FAILURE: Failed CMD: 400 

That's with the plane outside in my yard with full view of the sky.

marcmerlin, 2017-06-01:
@WickedShell ok, well, I'm just going to assume that I have a very crappy GPS and AP 3.8 may be more picky about it than AP 3.7.
The part I'm kind of worried about is that to override the GPS arming check, I had to disable '16' aka 'INS(INertial Sensors - accels & gyros)' and not '8' aka GPS Lock.
In other words, I have arming require 2+4+8+32+64+128+256+512 = 1006
This does not fill me with joy because I don't want to disable the check on gyros and accelerometers, but it seems that to disable the GPS check, I'm forced to.
http://ardupilot.org/plane/docs/parameters.html#arming-check-arm-checks-to-peform-bitmask

It's a new issue, so I've filed this instead https://github.com/ArduPilot/ardupilot/issues/6362

Back to the original point of this issue.
Yes, AP 3.8 fixes the ESC signal dropout I reported as part of this issue.
However since once I got 3.7.1 to spin my motor at boot, I thing we should potentially keep this issue open to fix 3.7.x

marcmerlin, 2017-06-01:
@WickedShell I ran out of time today, but I can look tomorrow.
Sigh, I wasn't really looking at downgrading again and do the double reflash but compass recalibration dance, but in the name of science :) ...

WickedShell, 2017-06-01:
@marcmerlin Yeah, as I said it's a bit of a pain... You could save your parameters after you did the compass calibration and just reload them though, that way you don't need to keep redoing the calibration tbough...

peterbarker, 2017-06-02:
On Thu, 1 Jun 2017, Marc MERLIN wrote:

> Are there instructions on how to flash "by hand" as you said? If it's going
> to be required for everyone upgrading on pixhawk and clones, if it can't be
> automated by the firmware, instructions on how to do it would be helpful

It ordinarily is automated; sometimes things don't work as expected :-)

I've put this document together:
https://github.com/ArduPilot/ardupilot_wiki/pull/827

marcmerlin, 2017-06-02:
Ok, so I think all the connex issues have been split in their relevant issues and are being tracked separately.
The only thing left for this issue is how 3.7 did spin my motor for a second, once while it was booting due to how it doesn't have positive control of the PWM signal during boot. 
I only had this once on one aircraft, so reproducing it may be hard, and this may be a very rare occurrence, but I'm thinking if the patch that fixes PWM dropping during boot is backported from the 3.8 branch, it would most likely address the issue I saw.

magicrub, 2017-06-02:
@marcmerlin is your BRD_SAFETYENABLE=0? Either way, I would think [this](https://github.com/ArduPilot/ardupilot/pull/4264) solves that problem. @WickedShell got any ideas on this? You've worked on the px4io recently... how could this happen?

marcmerlin, 2017-06-02:
@magicrub good guess, I do have BRD_SAFETYENABLE=0 in case I get an inflight reboot/power issue, I want the aircraft be recoverable, also I just don't seen the point (for me) of an arming button when I already have the simpler option of arming with the RC controller.
However, I have ARMING_RUDDER set to 2, so the aircraft isn't armed after it boots, but I can arm it by rudder, I just don't use/need the safety switch.
The motor therefore never runs after I boot the system, even at full throttle from the RC controller, since the aircraft isn't armed.
That said, the patch you posted is from last year, shouldn't it already be in 3.7.1?

magicrub, 2017-06-02:
yes, that fix is in v3.7. I fly without a switch too, I find it redundant butit does have a purpose.

The problem is some other external hardware system manages the PWMs, the PX4IO chip so there's a sync going on in the background. We talk high-speed serial to it and then it does the PWM in/out stuff for the first 8 channels. The idea is that's a simple chip, and  nothing should go wrong with it. less is better - code wise. If it doesn't get an update in a matter of a few hundred ms then it allows passthrough to allow manual flying during a flight controller reboot. But, it has to know which is the throttle so the FC sends a profile (and a mask) so it knows which channels it can override and which it should passthrough. The arming button actually connects to the PX4IO chip so the BRD_SAFETYENABLE=0 actually gives a command to the PX4IO to make it think it's always being pressed. Otherwise the FC throttle commands are ignored until the button is active.

marcmerlin, 2017-06-02:
I see, so you're saying it's an unfortunate working as it has to, even if not as desired.
Yes, it's very possible that one time I had the throttle active on the RC controller and it got passed through during boot.
However, I just tried to reproduce on 3.8 and it didn't happen, so maybe it does something better?

magicrub, 2017-06-02:
"maybe" is the m word..

marcmerlin, 2017-06-06:
Since there are talks of a possible 3.7.2, if that happens, backporting the throttle fix from 3.8 would probably be best (with the hope/assumption that it fixes the one time spurious motor start I saw with 3.7).
Or, if absolutely no one has seen this before, it's probably also acceptable to just close this issue as non reproducible in 3.7 and fixed in 3.8

magicrub, 2017-06-06:
I believe there was a lot added to the init and some was blocking causing this glitch. In v3.8 more has been pushed as async and things have been optimized so the init does not block.

magicrub, 2017-06-06:
I don't think this is something we can back port very easily because the optimizations are not clear... This is 9 months of changes were talking about.. I would suggest keeping v3.7.3 specific to the absolute critical stuff and then focus on getting v3.8 out.

magicrub, 2017-06-06:
@marcmerlin if you can confirm the problem is gone in v3.8 please consider this issue fixed and should be closed.

marcmerlin, 2017-06-06:
@magicrub I can confirm

closed 2017-06-06 by marcmerlin