ArduPilot/ardupilot #6264: How to get only GPS data as file ( such as .txt or .csv )  using telemetry?

YunJinHan, 2017-05-17:
#### Issue details
Hello I want to get pixhawk GPS data file using telemetry ...
I'm already done to show GPS log at APM planner 2.0 but how to extract pixhawk GPS data in file?
i want to get that Data as file ( such as .txt or .csv etc )
Is there any way to get GPS data as file ? 

#### Version
APM planer 2.0 and Mac OS

#### Platform
[  ] All
[  ] AntennaTracker
[ 0 ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
hex

#### Hardware type
pixhawk

OXINARF, 2017-05-17:
This issues list is for feature requests and confirmed bugs. For user support please post in the forum (http://discuss.ardupilot.org).

Your question is for GCS, Mission Planner, for example, can export logs to csv. I have no idea about APM Planner. You can post in the APM Planner category in the forum asking about it.

closed 2017-05-17 by OXINARF

YunJinHan, 2017-05-17:
I solved that problem using another mavlink open source.. thank you for answer me !

cafeclimber, 2017-09-14:
@YunJinHan Would you mind posting a link to the mavlink open source you used?

YunJinHan, 2017-09-15:
@cafeclimber  I used https://github.com/mavlink/mavlink protocol open source and https://github.com/YunJinHan/Sky_Hi_Jacker this project includes code to get gps data from pixhawk ( that's code link : https://github.com/YunJinHan/Sky_Hi_Jacker/tree/master/Get_Pixhawk_GPS )
Thanks

cafeclimber, 2017-09-15:
Thank you @YunJinHan This helped a ton