ArduPilot/ardupilot #6265: Bus definitons for ToshibaLED should be in HAL, not in

night-ghost, 2017-05-17:
ToshibaLED_I2C.cpp:

```
#define TOSHIBA_LED_I2C_ADDR 0x55    // default I2C bus address
#define TOSHIBA_LED_I2C_BUS_INTERNAL    0
#define TOSHIBA_LED_I2C_BUS_EXTERNAL    1
```

because that I2C bus on some hardware can be unused at all and pins it uses can be used for another needs, so if pin that used as SCL is low then all hangs forever in I2c::bus_reset().

OXINARF, 2017-05-17:
@night-ghost This is a more broad issue, we don't currently have a portable method to get the I2C buses that drivers should probe. This has been discussed in several places, but no conclusion was reached yet. I've changed the title of your issue so we can track this very important need.

night-ghost, 2017-05-17:
Is it not enough just to move definitions of TOSHIBA_LED_I2C_BUS_INTERNAL / INTERNAL to AP_HAL/board/smth.h and surround code in ToshibaLED_I2C.cpp by #ifdef - like it done for another I2C devices?

OXINARF, 2017-05-17:
@night-ghost No, because it's not just ToshibaLED, lots of drivers use I2C and need to know what buses should be probed - we can have an *I2C buses to probe* macro in HAL, but for Linux it would apparently not work well as the I2C bus number can change.

patrickpoirier51, 2018-11-17:
Just a reminder here,  there is still some work to do for the I2C bus scan as for example  here:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RangeFinder/RangeFinder.cpp#L647

On BBBMINI this is bus 2 ....so I have to patch manually until a proper bus scan work.  Please note that Bus Scan works fine for the PX4FLOW on the same bus