ArduPilot/ardupilot #6272: MAV_CMD_DO_MOUNT_CONTROL not changing yaw

Pedals2Paddles, 2017-05-18:
#### Issue details
The solo is sends mavlink mount control commands to ArduPilot. This is how it controls copter yaw and gimbal pitch in many of the smart shots. The companion computer interpolates gimbal pitch angles and copter yaw angles, then uses mavlink to command said movements.  This all worked fine on the old ArduPilot Solo firmware from the 3.2/3.3 days.  However on master the copter yaw response is no longer working. The gimbal pitch still responds, but not the copter yaw.  So we know the message/command is going through.  Just copter yaw is not responding.

* The companion computer uses `msg_mount_control` to send gimbal pitch and copter yaw.  This can be found in copter GCS_Mavlink here: https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L1904.
It then goes to the function in the mount backend found here: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mount/AP_Mount_Backend.cpp#L37
Where it then goes into the control function in the mount backend here: https://github.com/ArduPilot/ardupilot/blob/1297a7dedbac0883ee28e9cb36813262e0514f34/libraries/AP_Mount/AP_Mount_Backend.cpp#L45
Which finally concludes here in the mount backend:
https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/commands_logic.cpp#L1172

* I tried changing the solo's companion computer to use the newer `mav_cmd_do_mount_control` and the result is the same. Gimbal pitch responded, yaw did not respond. This one is in copter GCS_Mavlink here: https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L1227
Where it then goes directly into the same control function in the mount backend as the old one: https://github.com/ArduPilot/ardupilot/blob/1297a7dedbac0883ee28e9cb36813262e0514f34/libraries/AP_Mount/AP_Mount_Backend.cpp#L45
Which finally concludes here in the backend, again same as the other method:
https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/commands_logic.cpp#L1172

* I then setup and auto mission and tried using the  _Mission Command_ `do_mount_control`. That does actually work properly in all respects. It begins here in Copter commands_logic: https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/commands_logic.cpp#L111
Then goes down to the bottom here:
https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/commands_logic.cpp#L1172
And concludes here, which is where all the other mavlink commands also conclude:
https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/commands_logic.cpp#L1172

So this tells me at least from that last point onward at `set_angle_targets`, it works fine.  So something is not working properly in the Mavlink commands, or the code processing those mavlink commands which lead up to it. The result is again copter yaw not responding.  I don't see anything glaringly obvious, but I only know enough C++ and mavlink to break it. 

#### Version
3.5 / 3.6dev

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Solo, but likely any vehicle using mavlink mount control

#### Hardware type
Cube 2.1

#### Logs
Dataflash log attached. 
[Solo Mount Control.zip](https://github.com/ArduPilot/ardupilot/files/1009474/Solo.Mount.Control.zip)

Pedals2Paddles, 2017-05-18:
@tridge @rmackay9 this is what I talking about on the dev call. I did some further testing as you suggested to get to this point.

rmackay9, 2017-05-18:
Ok, this sounds to me like Copter thinks the Solo gimbal supports yaw so it thinks the main vehicle doesn't need to yaw.

Pedals2Paddles, 2017-05-18:
That would be way too easy for my luck!

Pedals2Paddles, 2017-05-22:
I put a `condition_yaw` command immediately following `mav_cmd_do_mount_control`, which is a working workaround now.  The `condition_yaw` command is working properly.  This doesn't appear to degrade the vehicle's performance. But it is a somewhat ugly workaround. It does however confirm the problem is not how the solo is calculating its yaw. 

This leads me to believe there is a problem in how ArduPilot is handling the mount_control message and command.

closed 2017-12-23 by rmackay9