ArduPilot/ardupilot #6279: Plane, SITL, ROS, mavros: Guided mode rejects nav commands - result 3

fbalaban, 2017-05-18:
#### Issue details
SITL with Plane instance. Trying to send commandLong, commandInt (through ROS-mavros) in guided mode but the message gets rejected (3). The command is a **MAV_CMD_NAV_WAYPOINT** with lat,lon.alt.

When sent to: SET_POSITION_TARGET_GLOBAL_INT, there is no response or log error/warning.

When trying to change mode to Guided (from Auto), it's done correctly. (so no connection or other issues)

Has been raised here: https://github.com/mavlink/mavros/issues/712

The code in [ArduPlane GCS_Mavlink.cpp L2064](https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/GCS_Mavlink.cpp#L2064) seems strange in the meaning that it doesn't handle lat,lon coordinates, like the same instance in Copter [does](https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L1693). Of course I can be wrong since I haven't studied the code thoroughly.

How a simple "change to guided - go to lat lon" stream behaviour can be accomplished?

#### Version

ArduPlane 3.6.0

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover

#### Airframe type

Fixed wing SITL Rascal

#### Hardware type
-
#### Logs
_Please provide a link to any relevant logs that show the issue_

khancyr, 2017-05-18:
Hello, it looks like set_position isn't support by plane.
To use MAV_CMD_NAV_WAYPOINT in GUIDED mode you need to set the current parameter to 2. see :https://github.com/ArduPilot/MAVProxy/blob/b2452edc85c6d8c83cc258e3e6219ac9b1268675/MAVProxy/modules/mavproxy_mode.py

fbalaban, 2017-05-18:
Hi, it doesn't seem to work.. 
I still get a "Command unsupported" response. 
BTW I've seen that at the link you've posted,  the "current" is after the command, but where is supposed to go in the [message definition](http://docs.ros.org/api/mavros_msgs/html/srv/CommandLong.html)? I know that after the command there are 7 parameters but in that script there are 9. I cannot find the mission_item_send definition..

WickedShell, 2018-08-26:
Try using COMMAND_INT with MAV_CMD_DO_REPOSITION (support is here: https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/GCS_Mavlink.cpp#L827 ). This is otherwise a support issue, if you need more help please go to discuss.ardupilot.org.

closed 2018-08-26 by WickedShell