ArduPilot/ardupilot #6283: waf: add upload target for Linux boards

lucasdemarchi, 2017-05-19:
I took a look on how to add an upload target (or option like it is in PX4), but I couldn't find an obvious way to do it.  My plan was:

`./waf build --targets bin/arducopter --upload`
This would implicitly call `./waf --destdir=<tmpdir> install` and then synchronize via rsync/scp the dir to a target IP, that could come from an env var

I can't find how to convince waf to add rule dependencies, though.  There are some examples in the manual, but I can't pass options like above. @guludo @tridge  any idea?

guludo, 2017-06-02:
@lucasdemarchi Hi. Sorry, I just saw this today. I've done something similar with a personal project. In my case, I made `waf` run install and then used `rsync` to send the install dir to a remote machine. I can take I look if you want..

guludo, 2017-06-02:
@lucasdemarchi Here is a fragment of my `wscript`:
```python
def configure(cfg)
    ... # code removed

    cfg.find_program('rsync', mandatory=False)

    if cfg.options.upload_dest:
        cfg.msg('Setting upload destination to', cfg.options.upload_dest)
        cfg.env.UPLOAD_DEST = cfg.options.upload_dest

def build(bld):
    if bld.cmd == 'upload':
        if 'RSYNC' not in bld.env:
            bld.fatal('rsync program seems not to be installed, can\'t upload')

    bld.recurse('src')

    if bld.cmd == 'upload':
        bld.add_group()
        if bld.options.upload_dest:
            bld.env.UPLOAD_DEST = bld.options.upload_dest
        if 'UPLOAD_DEST' not in bld.env:
            bld.fatal('Upload destination not defined. Either pass --upload-dest here or during configuration.')
        bld(
            rule='${RSYNC} -a ${bld.options.destdir}/ ${UPLOAD_DEST}',
            always=True,
        )

class UploadContext(Build.InstallContext):
    '''uploads the install directory using rsync. See "Upload options".'''
    cmd = 'upload'
```

guludo, 2017-06-02:
There should be a way to adapt this to the current code. Maybe I'll try it tomorrow... Time to get some sleep now :smiley:

lucasdemarchi, 2017-06-02:
Interesting. And yes, it's basically what I want. Thanks.

OXINARF, 2017-09-06:
This has been added with #6518

closed 2017-09-06 by OXINARF