ArduPilot/ardupilot #6285: Battery: support reading voltage of each cell

magicrub, 2017-05-19:
support reading voltage of each cell

magicrub, 2017-05-19:
Ping @marcmerlin

WickedShell, 2017-05-19:
@magicrub What is the issue for? This is capable with a SMBus battery, and there aren't enough ADC ports to do direct monitoring of any other battery cells?

lvale, 2017-05-19:
@WickedShell If we can have this with a SMBus battery then it is an incentive to get wider adoption of these.
This might pose other issues, like defining the current voltage settings on a per cell value and adding a new value with the number of cells

WickedShell, 2017-05-19:
@lvale yeah, it's done though :)

I don't know what you mean by current voltage settings per cell? Your SMBus device does the work for you,  you don't have to worry about it after that? I'm still confused what the issue is asking for :)

lvale, 2017-05-20:
This:

[image]

marcmerlin, 2017-05-20:
Also, it would be totally fantastic if one of the serial port on the pixhawk to be configured to be compatible with the frsky Sport 1 wire protocol.
We'd then get lots of cool and cheap sensors like
http://www.getfpv.com/frsky-flvss-lipo-voltage-sensor-with-smartport.html
and
http://www.getfpv.com/smart-port-rpm-sensor-with-2-temperature-sensors.html
The RPM sensor thing on AUX4/5 has never worked for me despite trying quite a bit, and it only works with certain ESCs. The frsky one works with any ESC while giving you 2 temps you can measure too.
But back to the lipo per cell sensor, you get that easily with their sensor on all your existing batteries.
Having to replace all our batteries just so that ardupilot can talk to them, seems like a no go for many when a simple $17 sensor takes care of that (and it even has a screen with readouts on it)
[image]

marcmerlin, 2017-05-22:
I opened a new issue to deal with reading Taranis sensors: https://github.com/ArduPilot/ardupilot/issues/1587
If it is solved, then in turn it would solve this issue easily.

OXINARF, 2017-05-22:
I still don't get what this issue is for. We already support reading the cell voltages where it is available. Supporting a new battery monitor is fine, but that's not what this issue says - and I guess #6296 already covers that request.

WickedShell, 2017-05-24:
Closing this in favor of #6296 as it more clearly describes the top level issue, and the core things that need to be done to read that sensor.

closed 2017-05-24 by WickedShell