ArduPilot/ardupilot #6287: DDFP tail rotor starts without arming flight controller

ChristopherOlson, 2017-05-21:
#### Issue details
Direct Drive Fixed Pitch tail rotor is active and starts without arming flight controller

#### Version
3.4.6 and 3.5rc's

#### Platform
[  ] All
[  ] AntennaTracker
[x  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Helicopter

#### Hardware type
Pixhawk

#### Logs
Video demonstrating the issue:
https://youtu.be/FoWk5IXjB1g

auturgy, 2017-09-15:
What is your BRD_SAFETY_MASK param value?

CraigElder, 2017-09-15:
http://ardupilot.org/copter/docs/parameters.html#brd-safety-mask-channels-to-which-ignore-the-safety-switch-state

ChristopherOlson, 2017-09-15:
The DDFP tail function worked fine in 3.3.3. It got broken in 3.4. The problem is in the motor library for heli, not the board safety bitmask. I fixed it in my fork for 3.4.6 but the code changed again for 3.5. So I'll re-do the fix for 3.5, flight test it, and PR it to 3.5 and master. I had forgotten I left this issue open so I'll close it.

R-Lefebvre, 2017-09-15:
Chris, please leave this issue open.  It is an issue in 3.4/5 still.

This has *nothing* to do with BRD_SAFETY_MASK.  The motors should never turn until the system is *Armed*.  This worked properly in 3.3, but was broken in 3.4, likely due to changes with the 0-1 inputs.

The math for the DDFP calculation is also comletely wrong.  The fix is something like this:

https://github.com/R-Lefebvre/ardupilot/commit/cbad27106f1a56b215bd9fd0b18092edd1212b24

This is not finished.  It still has the motor able to start if Ch8/Interlock is raised, but is not Armed.  It needs to check if motors are armed, otherwise output 0.

ChristopherOlson, 2017-09-15:
Rob, ok - I'll leave it open. I did refine the fix for 3.4.6 and then the next release came out and was totally different again. So I forgot it and never got around to looking at it again.

tridge, 2017-09-20:
@ChristopherOlson thanks for the explanation. Can you give a link here to the fix you did for 3.4.6?

tridge, 2017-09-20:
> he math for the DDFP calculation is also comletely wrong. The fix is something like this:
> R-Lefebvre/ardupilot@cbad271

that looks good to me, apart from the issue you mention

> This is not finished. It still has the motor able to start if Ch8/Interlock is raised, but is not Armed. It needs to check if motors are armed, otherwise output 0.

maybe we should have a motor_running() boolean method on AP_MotorsHeli_RSC and check that?

ChristopherOlson, 2017-09-20:
@tridge it's the same thing Rob linked to. Before I did this the tail rotor would start unarmed if you picked up the helicopter and yawed it. This fixed that problem so it wouldn't start without channel 8 high. But it still starts without arming. That's why I never pushed it (until now) to my github fork

https://github.com/ChristopherOlson/ardupilot/commit/2797828ceb7a519f216808d7d3640a8768220b5a

tridge, 2017-09-20:
I've done a PR here that fixes it:
PR #6983 
can you take a look Chris and test it? I've only tested in SITL

ChristopherOlson, 2017-09-20:
yes, thanks tridge. I will build it and test it to make sure it works properly.

ChristopherOlson, 2017-09-21:
@tridge should I do the test build and test flights with master or the current release of 3.5? It shouldn't make any difference but wanted to ask before I do it.

tridge, 2017-09-21:
@ChristopherOlson either would be fine. Let me know if you'd like me to make a branch with this change on 3.5

ChristopherOlson, 2017-09-22:
@tridge I have a branch on my fork:
https://github.com/ChristopherOlson/ardupilot/tree/pr-heli-DDFP-tail-fix

I built the change on AC3.5.3 and it passes all the bench tests. If anyone else wants to test it, the custom build is here:
https://drive.google.com/drive/folders/0B5oLpNzxih4tZHhsOTFqRl9TUzA?usp=sharing

I will proceed with flight tests tomorrow in my Trex 500. The change was minor and I have flown the code before with 3.4.6 so I don't expect any issues with tail control in flight testing.

ChristopherOlson, 2017-09-23:
@tridge I flew this patch today and it works fine. Tail rotor does not start unless the flight controller is armed, and it starts and stops immediately with the HeliRSC control. The math works fine and no issues with tail control. The tail rotor is again disabled even with HeliRSC high after disarming.

Tested with the build linked above on AC3.5.3. As far as I'm concerned this issue can be closed unless someone else has any objections.

closed 2017-09-23 by ChristopherOlson