ArduPilot/ardupilot #6295: Support for crow flaps

marcmerlin, 2017-05-22:
writing down a quick chat I had with @magicrub 

Crow flaps are absolutely essential for airframes that float well, almost too well too land.
First, crow flaps are ailerons up, flaps down. Note that flaps are often set up so that flap neutral is with servo at 90 or 95% since flaps do not need to go up, but need to go down a lot.
Example of how it looks like:
https://youtu.be/F6d4VKdc34A?t=16m25s

Why? Because this allows dropping a lot of altitude without picking up a lot of airspeed without needing a reversible prop/ESC

How?
On my setup, the bottom half of the throttle stick controls the amount of airbrakes (i.e. crow flaps) you get. The top half of the throttle spins up the motor and control how fast it turns.
Going from one mode (crow flaps) to power should not require a switch, this allows for easy transition from air brakes to power for go around, by simply moving the throttle.
There is honestly no reason to ever spin up the motor while crow flaps are active.

Currently, doing this is messy. I wrote a doc here for flaperons
http://marc.merlins.org/perso/rc/post_2015-08-04_Using-Flaperons-With-Ardupilot.html and the as hinted on  http://ardupilot.org/plane/docs/fixed-wing-faq.html#how-would-i-setup-crow-flaps I think you setup the same flaps channel to output to flaps in opposite direction from the ailerons/flaperons.
However, that doesn't really work I believe since AP wouldn't be able to actuate the flaps separately from the flaperons

Workaround:
For now, since I can't get it to work through AP, I switch to manual for landings and control crow flaps directly from the RC remote, bypassing any AP help. Sadly it means I also lose airspeed/stall protection. 

Work:
AP needs to know that flaps can be regular flaps (with a setup where subtrim is such that flaps can go 5% up and 95% down with servo center being around 45% down) and at the same time have a mode to lose altitude without picking up airspeed where ailerons are actuated as reverse flaperons, and flaps move at the same time.
For extra points, if the  aircraft is in FBWA/STAB, it would allow having 50% of the throttle stick do crowflaps/airbrakes and the other 50% do the motor, while in FBWB/CRUISE/AUTO it would actuate crow flaps on its own if desired.
I think the ultimate high angle descent landing would do an aggressive descent with full crow flaps and then switch to half flaps to allow pitching the nose up and doing a slower airspeed landing.

marcmerlin, 2017-05-22:
For (back ) reference, this was the original discussion on how to setup flaperons: https://github.com/ArduPilot/ardupilot/issues/2524
As a warning to others, on the airframe I tried it on (X-UAV Talon), setting full flap(eron)s down with a rotating switch took effect after a 1 second delay (some default AP uses to actuate flaps slowly). When they were fully down, it violently pitched the airframe nose down, and the delay it took to bring the flaps down to 0 was enough for the airspeed and Gs to be enough when the nose was pulled up, to unglue a ruddervator, causing an uncontrolled high speed crash into the ground.
So basically be very careful when you're testing AP controlled flap(eron)s and be wary of the delay that's there by default when flaps are activated and raised.
This will obviously apply even more when testing flaperons if they are not exactly in sync, and being aware of the nose up/stall tendency they can cause (I have a mix with elevator down on them). FBWA with an airspeed sensor should compensate, but if you don't have an airspeed sensor with stall protection, beware.

Naterater, 2019-01-04:
Crow flaps are now available in master, so I think this issue is solved.

IamPete1, 2020-07-28:
This close the issue I think [Differential Spoilers & Full House Wing](https://ardupilot.org/plane/docs/differential-spoilers.html#differential-spoiler-options)

closed 2020-07-28 by IamPete1