ArduPilot/ardupilot #6296: Support Taranis Smart Port sensors as data sources for pixhawk/ardupilot

marcmerlin, 2017-05-22:
While I personally care more about planes, this really would be useful to all vehicles. @magicrub this is what we discussed in person, just writing it down so it doesn't get lost.
Sadly, 18 months ago instead of filing a request I actually tried to do this myself, but while I had some success listening on an S-Port bus via this method
[image]
(except on the left I used a serial-USB converted and sent it to my laptop), I used 3 serial to USB converters and 2 of them at 57600bps never got the signal, while the 3rd one was tuned slightly differently and was able to get the data, but I never was able to generate a packet to interrogate a sensor so that it would return something (I had to put an X8R on the bus and rely on it to interrogate the sensors and listen passively).
I'm opening this so that others can have a look.

To remove confusion,  https://github.com/athertop/MavLink_FrSkySPort/wiki 
is great, but it's meant to send your ardupilot telemetry through a Taranis radio link and display it on your remote control.
This request is totally different, it is to leverage some frsky sensors and use them as data sources for ardupilot. For 3 reasons:
1) some of that data could be used directly by ardupilot
2) some can be passed via mavlink to be displayed by an OSD and/or logged on the ground
3) this also allows using Taranis sensors on a non Taranis link (actually in my case I have a Taranis but I use LRS, so I can't use the sensors either).

The ones I'm thinking of are:
http://alofthobbies.com/frsky-sp-flvs-smart-port-lipo-voltage-sensor.html
[image]
=> this is mostly useful for people who would like to monitor per cell voltage for batteries that are not balanced. This is relevant to https://github.com/ArduPilot/ardupilot/issues/6285

http://alofthobbies.com/frsky-sp-rpm-smart-port-rpm-and-dual-temperature-sensor.html
[image]
This one is much more useful in my opinion: this allows adding an RPM sensor to any existing motor and ESC. Even in my case I have a castle ESC with RPM signal out and never managed to get it work with the RPM sensor feature on AUX4/AUX5 that was added in ardupilot, but with this frsky sensor, I get RPM right away, no problems.

marcmerlin, 2017-05-25:
@magicrub you mentioned to me that you knew someone working on frsky sport libraries. Is this feature request something that would be interesting to them?

lucidwan, 2017-05-25:
It would be nice if this could be supported as well. Much simpler and accurate current sensing solution.
[http://www.frsky-rc.com/product/pro.php?pro_id=133](url)

magicrub, 2017-05-25:
@marcmerlin I was mainly thinking of @floaledm from Craft & Theory.

floaledm, 2017-05-27:
Thanks Tom for referring this to us. I think it's an excellent suggestion that we would like to see implemented as well, especially considering the work done recently on the Smart Battery lib. This will require some adjustments to the FrSky lib. ArduPilot should likely first detect if a receiver is hooked onto the Smart Port bus, and if not, have ArduPilot emulate the polling requests of the receiver (since sensors only provide data after being polled).

geofrancis, 2017-08-14:
the sensors i would like to see supported: 
FLVSS/MLVSS for per cell monitoring
FAS40/FAS150s for current monitoring
ASS100 for airspeed

and finally this DIY flow sensor 
https://www.rcgroups.com/forums/showthread.php?2906155-Fuel-flow-sensor-for-FrSky-S-Port-receivers

then you could get accuate fuel consumption figures for petrol/glow engines, if you just mapped 1ml=1mah then it should work with minimal confiuration since you could still use the capacity failsafes and warnings.

auturgy, 2017-11-24:
Just a little bump, as there has been another request, and also a new FrSky protocol that may make it easier to implement (its uninverted serial at 115200).
I've started a discuss thread here: https://discuss.ardupilot.org/t/frsky-sensor-support/23431 to gauge demand.

Voidminded, 2018-01-20:
I really like (and need) this idea. I'm specially interested in the RPM sensor (https://alofthobbies.com/frsky-sp-rpm-smart-port-rpm-and-dual-temperature-sensor.html), and am willing to spend some time to develop this feature. 
The only question here is: does anyone have any idea what is the data providing frequency and latency of these sensors ? I need high frequency and preferably real-time RPM data.

Szpero, 2018-04-17:
Bump. I'm looking forward for so such functionality too.
Now my 12" quad moving to Crossfire link however FLVSS will be not in use. I wish to have my cell readings back again to see battery condition during the flight.
Maybe this thread will help a little (frsky library):
https://www.rcgroups.com/forums/showthread.php?2245978-FrSky-S-Port-telemetry-library-easy-to-use-and-configurable

ChrisBird, 2019-01-17:
Once I'm done on the failing airspeed sensor pr I'll give this one a shot.  I have some some of the sesnors and wouldnt mind having that data available.

floaledm, 2019-01-17:
Hi Chris,
Good to hear. I'd be glad to help with the review of such PR.
Once it's done for one FrSky sensor (personally, I think that the FLVSS/MLVSS sensor support would be most beneficial to the community, but you do you; the FrSky airspeed sensor - aka ASS =) - would provide another economical hardware option), I can arrange for Craft and Theory to send you several others (e.g., current, RPM) if you need them for dev. I can otherwise assist in some testing...

ChrisBird, 2019-01-18:
@floaledm thats would be great let me get the mlvss and the voltage one done first which i have already.  Then we can discuss if i need any extra hardware further to do the others.  Will be starting it tonight.

floaledm, 2019-03-27:
I would like to relay what @juichi emailed me which elegantly summarizes what is required for this functionality if we want to do it properly:
- Flight controller acts as a bus controller for S. Port - polling all available sensors and treating the data as an input to the flight controller (no frsky rx)
- Flight controller acts as a bus monitor for S. Port - watches the receiver polling on the S. Port bus and gathers sensor data (frsky rx, no passthrough tlm)
- Flight controller acts as a remote terminal + bus monitor for S. Port - can see all the traffic on the bus (gathers sensor data) but will also act as an S. Port sensor as in passthrough tlm

If I may add, in that last mode, a remote controller using OpenTX could very well communicate with the flight controller, and perform tasks such as polling or setting downstream data rates (resulting in a configurable S.Port protocol in ArduPilot), or even perform ArduPilot parameter configuration using that remote, and without the need for a ground station (similar to what is done in CleanFlight/BetaFlight).

ChrisBird, 2020-06-11:
PR finally raised for this, sorry guys for the long delay.  In the end an $15 logic analyser unblocked this from happening.

https://github.com/ArduPilot/ardupilot/pull/14583