ArduPilot/ardupilot #6297: AC_PrecLand _inertial_history buffer should be set by parameter

fnoop, 2017-05-22:
#### Issue details
In AC_PrecLand, the _inertial_history buffer effectively determines the latency of the incoming landing coordinates to the IMU for attitude correction.   This is set to an integer multiple of the timing loop (400hz) and is fixed in AC_PrecLand.h at 8, which corresponds to 20ms.
`    AP_Buffer<inertial_data_frame_s,8>       _inertial_history;`

This figures was apparently derived from the latency of the IRLock system which has specific dedicated hardware for landing detection.  However, for more general use the 'Companion Computer' backend is used to send Mavlink messages through to AC_Precland.  The latency of these messages from the camera capture will vary widely, but will almost always be far higher than the preset 20ms.  This leads to incorrect IMU correction and dangerous movements of the craft.

The EKF estimator (the default) seems to be particularly sensitive to this, being largely unuseable and quite dangerous when using the Companion backend.

#### Version
3.5-rc6

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quadcopter

#### Hardware type
Pixhawk

OXINARF, 2017-05-22:
We can do this, but I don't think it is the correct approach. The buffer should be big enough to support the delay with a CC backend, but it shouldn't be arbitrarily big because if the delay is too big the data will be near zero-value. Also, this doesn't fix the fundamental problem: there is no correlation of the data coming from backends and the IMU data.

patrickpoirier51, 2017-05-22:
Most of the CC based camera tracking are lagging with a latency between 80 to 200 millisec. 
I agree with the fact that a the best approach is to correlate the camera pose with the IMU (either camera based as in VIO or OpticalFlow or time synchronization with the internal IMU on each frame), but in any case we would still  need some kind of ''fine adjust'' for the delay;  just like we have with the OpticalFlow (using the delay parameter within EK2 & EK3), than can add an extra 200 millisec within the sensor fusion estimator loop.

fnoop, 2017-05-22:
@OXINARF Agreed it doesn't fix the fundamental problem of data correlation, but that doesn't look like it's going to happen anytime soon.  In the meantime, we're left with something that doesn't work well, or at all for the majority of cases, and is dangerous at worst.  I'm just trying to suggest minimal incremental improvements - quick hits - to mitigate these issues.
Given that the raw estimator seems to be more forgiving than the EKF, I wonder if this should be the default, but that possibly belongs in a different issue.

OXINARF, 2017-05-23:
Adding a delay parameter is fine, although I think that using timestamps would probably be better since CC will probably not have a constant delay. We can use that delay parameter to determine the size of the buffer and correlate the data.

peterbarker, 2021-07-30:
This is now calculated:
```
 // calculate inertial buffer size from lag and minimum of main loop rate and update_rate_hz argument
    const uint16_t inertial_buffer_size = MAX((uint16_t)roundf(_lag * MIN(update_rate_hz, AP::scheduler().get_loop_rate_hz())), 1);

```

closed 2021-07-30 by peterbarker