ArduPilot/ardupilot #6300: AP: Crash, partial data freeze, during unusual autotune procedure.

AndKe, 2017-05-22:
Forum user posted this log:
http://diydrones.com/forum/topics/help-analyse-the-reason-behind-crash?commentId=705844%3AComment%3A2379548&xg_source=msg_com_forum

AP 3.7.1  in autotune, using rudder input while flying otherwise ok, even RTL. 
- he gets a throttle FS,(AP switches to circle) , RCOUT is frozen, altitude and other data looks very strange too, ~70sec later the plane crashed. - and suddenly updated altitude from 140 to 1.5 m

OXINARF, 2017-05-22:
This issues list is for feature requests and confirmed bugs. For user support please post in the forum (http://discuss.ardupilot.org).

@AndKe I understand it is a weird log, but it needs to be posted on the forum for analysis first.

AndKe, 2017-05-22:
@OXINARF , as you may not have seen, the issue was posted on dyiforums (maybe that don't qualify) - I analyzed it as far I could, and concluded that it needed attention from somebody better than me, as it looks like a nasty bug.   
So is it wrong forum ?  - or is the analysis on forum not good enough ?

kd0aij, 2017-05-22:
It's interesting that there's a 70 second dropout in the log, and that PM.MaxT is reported as 70,860,154 usec after the log resumes. What might cause the main loop to hang for that long?

AndKe, 2017-05-22:
good to see it attracts some interest, that's why I think the mystery and severity deserves more than a forum.

kd0aij, 2017-05-22:
link to log: https://www.dropbox.com/s/repd6lq1n815dc5/75.BIN?dl=0

magicrub, 2017-05-22:
This log has some serious questions that need answers. Firstly, the main processor definitely locked up somehow but became unstuck. Problem is Plane v3.7.x is old, even though it's the current release, so the chances are high that it has been fixed but this is something that we need to verify this particular problem has been fixed by all the scheduler changes and everything else that has happened in the past 9 months.

magicrub, 2017-05-22:
Here is the interesting bit in the .LOG version:

...
MSG, 2070011167, Throttle failsafe on 1817
MSG, 2070011319, Failsafe. Short event on,
MODE, 2070011388, CIRCLE, 1, 0
MSG, 2070011409, Flight mode = 1
... (more normal log stuff, and then 500ms later..)
AHR2, 2070592331, 6.05, 10.27, 108.97, 185.17, 32.4796916, 44.1861767
POS, 2070592356, 32.4797891, 44.1865879, 164.48, 0
IMU, 2070610829, -0.09703262, 0.09053399, 0.171757, -1.61282, -2.096814, -11.71483, 0, 0, 56.36288, 1, 1
IMU2, 2070610829, -0.1294468, 0.1121855, 0.1910543, -1.777946, -1.824077, -11.16479, 0, 0, 59.25, 1, 1
SONR, 2070611243, 0, 0, 0, 0
RFND, 2070611252, 0.00, 0.00
IMU, 2141472689, 0.0001642299, -0.008098508, 0.00637598, -2.10142, 2.703976, -9.541997, 0, 0, 62.70083, 1, 1
IMU2, 2141472689, -0.07486791, 0.001776762, 0.01980499, -2.267078, 2.858707, -9.067234, 0, 0, 65.375, 1, 1
GPS, 2141473400, 4, 487300200, 1949, 12, 0.74, 32.4797886, 44.1865877, 163.17, 10.69002, 89.8928, 2.58, 1

Note the time stamp jumps from 2070.611252s to 2141.472689s (71s gap).

Of note, normally a 5Hz GPS has a timestamp with exact 400ms interval. The GPS msg timetsamp goes from 487299800, then the long freeze, then 487300200 (delta 400ms) and then 487370800 (delta 70600ms) which means, among many things, it held it's state through the lockup. It could also be explained by the sdcard dataflash buffer being queued during the lock up too. None of which is good news but interesting clues.

tridge, 2017-05-22:
@AndKe can you try and find the tlog as well?

AndKe, 2017-05-22:
@tridge I asked the user to post .tlog - I cannot help with finding it as the operator is a random forum user I  were just trying to help.

magicrub, 2017-05-23:
Looks like a power failure. Power overload, not a software problem. Closing.

closed 2017-05-23 by magicrub