ArduPilot/ardupilot #6306: SET_MODE(Manual) from MAVlink doesn't check if there's RC input Failsafe

suraqis, 2017-05-23:
#### Issue details
When flying out of RC transmitter signals range and then tried to change flight mode from Auto or RTL to Manual from GCS (Mission planner) via Action tab then Arduplot doesn't check if there's RC failsafe and enters Manual mode with no actual RC input and doesn't detect situation as a failsafe anymore and not switching to Circle and RTL as configured.

#### Version
Arduplane 3.7.1

#### Platform
[ ? ] All
[ ? ] AntennaTracker
[ ? ] Copter
[ x ] Plane
[ ? ] Rover

#### Airframe type
Fixed wing with V-tail

#### Hardware type
Pixhawk 

#### Logs
Easily reproduced

OXINARF, 2017-05-24:
In Copter this is the intended behaviour - just because you lost RC connection it doesn't mean you can't control your vehicle with your GCS. In Copter we do have a macro (disabled by default) to disallow mode changes when in RC failsafe:  https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L847

@tridge @magicrub is this the intended behaviour for Plane?

suraqis, 2017-05-25:
I think it can't be an intended behavior for plane because we had a crash due to this. Switch to Manua wasl  caused by freezes of Mission Planner when link quality is not good enough. While MP freeze dropdown menu with flight modes can spontaneously close with Manual mode selected provoking erroneous action of operator. Flying Plane in Manual require very responsive control link and I'm not sure if joystick control via GCS is responsive enough. So I think entering Manual mode on Plane need to be checked for RC input.

magicrub, 2017-05-25:
Plane does not perform any checks when switching to any mode, every mode change request is successful. This is different than copter which will fail in certain circumstances. For example, you can switch to AUTO without a GPS lock in plane but not in copter. There are no additional checks for a GCS/MAVLink request.

I see your point and agree that it could be improved for cases just like this.

WickedShell, 2017-05-25:
This is indeed the intended behavior on plane. Mission Planner will prompt you before issuing mode changes so that should be handled, but using mode manual (also cruise/fbw) has real world use cases without inputs. (The former can be used to lock servos in place/stop trying to move, or depending upon your RC receiver cause an aerodynamic termination, the latter can be used to glide to the ground in a stabilized manner).

Now I agree that things like mode switch changes (IE RC commanded mode changes) should sanity check that manual input is present, but I do rely on the described plane behavior (less so now then I used to). I could see controlling it with a param, but I don't think we can do away with it.

suraqis, 2017-05-25:
There's two different things in that crash: 1) SET_MODE doesn't check some conditions and 2) manual mode doesn't check RC failsafe once already detected it successfully. So the problem was not only in set_mode Manual without checks, but in entering Manual mode without RC Failsafe reset. Maybe this behavior is more consistent.

WickedShell, 2017-05-25:
Again, I actually leverage that behavior and think of it is a good thing. It sounds like the only thing that got you into a problem was a GCS initiated error.

OXINARF, 2017-05-26:
> Switch to Manua wasl caused by freezes of Mission Planner when link quality is not good enough.

@suraqis This shows that the issue you have is with MP and not with the way Plane operates.

It looks like @WickedShell is against changing this behavior, but @magicrub is willing to change it. I think I'll ask @tridge to give the decisive vote :slightly_smiling_face:

magicrub, 2017-05-26:
I'm on the fence about it.

WickedShell, 2017-05-30:
This was discussed on the dev call today, the consensus of the call is that this is actually intended/desired behavior in a lot of emergency situations. The problem was encountered through a GCS initiated mode change, the GCS could be checking that you don't have RC input and should be prompting you to confirm that you want to change mode first. The current preference is to address this via having the GCS sanity check before requesting the mode. IE https://github.com/ArduPilot/MissionPlanner/issues/1560

closed 2017-05-30 by WickedShell