ArduPilot/ardupilot #6307: Ardupilot doesn't check if active WP number is greater than actual mission total WP number when switched to Auto

suraqis, 2017-05-23:
#### Issue details
There's sequence:
1) "Long mission" (let's assume 20 WPs) loaded to Ardupilot
2) Mission proceeded to any great enough WP number, let it be 15
3) Mission canceled with RTL with active WP number 16
4) New "short mission" (let's assume 10 WP) loaded to autopilot from Mission Planner while flying in RTL. There's no 16th WP in mission
5) Change mode to Auto commanded from Mission Planner
and then erroneous
6) Ardupilot fly to WP 16 from previous "Long mission"
7) Switch to RTL after WP 16 reached

#### Version
Arduplane 3.7.1

#### Platform
[ ? ] All
[ ? ] AntennaTracker
[ ? ] Copter
[ + ] Plane
[ ? ] Rover

#### Airframe type
Fixed wing

#### Hardware type
Pixhawk

#### Logs
Easily reproduced with SITL

magicrub, 2017-05-23:
What are you expecting the behavior to be?

magicrub, 2017-05-23:
This is a good starting point causing the current behavior:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L415

WickedShell, 2017-05-23:
It shouldn't resume into an old mission like that, should have gone RTL. If
no one beats me to it it's on my plate for tomorrow...

suraqis, 2017-05-23:
I think it would be good to reset mission active WP number with new mission loaded

magicrub, 2017-05-23:
when switching modes it will call start or resume which will eventually [hit this line](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mission/AP_Mission.cpp#L97) which should take care of this scenario because it fetches the mission item from flash memory at the index which was previously truncated to (the last index).

WickedShell, 2017-05-23:
@magicrub I agree that on first glance it looks fine, I have to recreate the orginal report in SITL still, but it shouldn't go back to a item beyond the mission length when resuming. (Honestly thought I had test this at one point, but I can't recall, will test tommorow and either reproduce/chase it down or close it at that point :) ).

@suraqis We don't want to reset the mission number when a new mission is loaded, because its undefined when that has actually finished loading. Uploading missions (or rally points, or parameters) aren't a transaction system where we copy over everything at once, so there isn't any clear time as to when things would be reset. I do agree that on resuming auto though that it shouldn't go beyond the end of the mission.

WickedShell, 2017-05-24:
Fixed in #6314

closed 2017-05-24 by WickedShell