ArduPilot/ardupilot #6309: Replay Tool compilation fails

clement-joly, 2017-05-24:
#### Issue details
When trying to build the EKF Replay Tool, the build fails with the following error messages:

"[335/653] Compiling libraries/AP_InertialSensor/AP_InertialSensor.cpp
In file included from ../../Tools/Replay/LR_MsgHandler.h:3:0,
                 from ../../Tools/Replay/LR_MsgHandler.cpp:1:
../../Tools/Replay/MsgHandler.h:88:5: error: expected unqualified-id before ‘[’ token
compilation terminated due to -Wfatal-errors."

This seems to be due to the "[[noreturn]]" in MsgHandler.h. I've tried commenting out noreturn to get past that issue but then I run in another compilation error:

"[330/653] Compiling Tools/Replay/Replay.cpp
../../Tools/Replay/LR_MsgHandler.cpp: In member function ‘virtual void LR_MsgHandler_PM::process_message(uint8_t*)’:
../../Tools/Replay/LR_MsgHandler.cpp:453:93: warning: format ‘%lu’ expects argument of type ‘long unsigned int’, but argument 3 has type ‘uint64_t {aka long long unsigned int}’ [-Wformat]
../../Tools/Replay/LR_MsgHandler.cpp: In member function ‘void LR_MsgHandler_GPS_Base::update_from_msg_gps(uint8_t, uint8_t*)’:
../../Tools/Replay/LR_MsgHandler.cpp:180:20: error: ‘GMS’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
compilation terminated due to -Wfatal-errors.
cc1plus: some warnings being treated as errors"

Any thoughts on how to fix this?

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

OXINARF, 2017-05-24:
What compiler are you using?

clement-joly, 2017-05-24:
Beautiful! I've just upgraded to gcc 6.2.0 and Replay compiled (I was using gcc 4.7.3 previously).
I'm closing the issue.
Thanks for the help!

closed 2017-05-24 by clement-joly