ArduPilot/ardupilot #6317: ALTITUDE: support reading altitude of copter

Ahrovan, 2017-05-24:
#### Issue details
pixhawk.ethz.ch/mavlink/#ALTITUDE contain the current system altitude and useful for use in ros (and ...). please Introduce a way to solve this problem and add reading altitude of copter.

#### Version
_ardupilot master,Copter,pixhawk_

amilcarlucas, 2017-10-22:
This fix is easy to do. This is a good first issue for a student starting to work in the codebase. Unfortunatly we do not have a label for that.

magicrub, 2017-10-22:
Altitude is available in about a dozen different MAVLink packets. Adding this particular one would be good, but you can just look at others for now to get you going

IamPete1, 2021-08-15:
https://mavlink.io/en/messages/common.html#ALTITUDE

aliakbarkhan, 2022-03-06:
> #### Issue details
> pixhawk.ethz.ch/mavlink/#ALTITUDE contain the current system altitude and useful for use in ros (and ...). please Introduce a way to solve this problem and add reading altitude of copter.
> 
> #### Version
> _ardupilot master,Copter,pixhawk_

is this issue still open i would like to work on it

amilcarlucas, 2022-03-07:
Is still open, please do.

tzarjakob, 2023-02-24:
This issue is solved, the message was added [here](https://github.com/ArduPilot/mavlink/blob/734d04188897f70c8b19e119543a395d4f61d6de/message_definitions/v1.0/common.xml#L5070)

IamPete1, 2023-02-24:
@tzarjakob although it made it to the AP MAVLink repo we don't actually send it yet.

tzarjakob, 2023-02-26:
That's true, I have made a patch for supporting this message.

peterbarker, 2023-02-26:
Oops.  Didn't reference this issue on my PR: https://github.com/ArduPilot/ardupilot/pull/21652