ArduPilot/ardupilot #6322: Copter: vehicle may backup if mission with takeoff as first command is initiated

rmackay9, 2017-05-25:
Try this in the simulator:

- set up a mission with a take-off command to 50m followed by a waypoint command
- set MIS_RESUME = 0
- arm and takeoff in stabilize or loiter, fly the vehicle forward at a reasonably fast speed but do not climb above 50m
- initiate AUTO
- the take-off command sets a horizontal target based on the vehicle's position at the moment the take-off command was initiated.  If the vehicle is moving horizontally it will continue flying forward for a bit but then slow and return to where it was.

The solution is that if the vehicle is not-landed, the take-off command should use the horizontal stopping point as it's target instead of the current horizontal position.

codeoftherings, 2018-05-21:
Can I take up this as my first job here?

rmackay9, 2018-05-22:
@codeoftherings, sure, give it a try!

codeoftherings, 2018-05-22:
@rmackay9 can you tell me the exact libraries that are causing this bug?

khanasif786, 2022-01-08:
@rmackay9 could we take the location of first waypoint (other than takeoff) as the setpoint.

haaaaaaarshs, 2026-10-26:
Hey, I would like to work on this issue. I'm new to ArduPilot, so I'll start by setting up Copter SITL on my Mac and trying the reported steps on the current master branch. I'll share an update by Tuesday, October 27th with what I find, including any setup problems.

Before getting started on a fix, is the horizontal stopping point still the intended target? I saw the later suggestion about using the first waypoint, so I wanted to clarify the expected behavior before making changes.

haaaaaaarshs, 2026-10-30:
Hey, sorry for the delayed update. I ran into a disk-space issue while setting up copter SITL on my Mac. After freeing some space, I installed the dependencies and built the Copter executable at commit 72b238201f. I've checked that it launched and displays its help output, but I haven't run the reported flight steps yet.
I'm still working through the remaining storage issue, then I'll try the reproduction steps and share what I find.