ArduPilot/ardupilot #6323: AP_BattMonitor: Unrealistic high ampere values

tbnobody, 2017-05-25:
#### Issue details
The Battery Monitor shows absolutly unrealistic Current (Ampere) values. With firmware 3.4.6 everything is fine (idle current ~600mA). When installing 3.5-rc7 the value changes to 10-20A! The voltages seem to be ok. 

My Config is:
```
BATT_AMP_OFFSET: 0
BATT_AMP_PERVOLT: 0,7023295
BATT_CURR_PIN: 3
BATT_VOLT_PIN: 2
BATT_MONITOR: 4
```
If I go to Mission Planner --> Initial Setup --> Optional Hardware --> Battery Monitor I see the following: 
Version 3.4.6: A valid double value (~0,6)
Version 3.5-rc7: Just a integer value (mostly 0)

Even if I change the BATT_AMP_PERVOLT to something smaller or larger I don't see any double values in the textfield mentioned above.

#### Version
3.5-rc7

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quad

#### Hardware type
Pixhawk

tbnobody, 2017-05-25:
To qualify this issue, I tried 3.5-rc4 (current is shown correct) and 3.5-rc5 (current is NOT shown correct).

magicrub, 2017-05-25:
oh, well that's certainly a useful datapoint! Thank you!

mtbsteve, 2017-05-25:
Have you the latest MP beta installed?
There is a known defect in MP in combination with AC3.5 which caused unrealistically high amp values

tbnobody, 2017-05-25:
Grml. You are so right, and I am such a fool. 
The calculation the current was changed: https://github.com/ArduPilot/MissionPlanner/commit/aa1d0418c406b1bcd685e5bf5b9f9d876d5ff5a4

I am closing this issue.

closed 2017-05-25 by tbnobody

OXINARF, 2017-05-26:
@tbnobody This issues list is for feature requests and confirmed bugs. For user support please post in the forum (http://discuss.ardupilot.org). That's even more requested when testing beta versions - actually if you had searched the forum first you would have probably found your answer :slightly_smiling_face: