ArduPilot/ardupilot #6352: Copter-3.5: OLED display always says EKF:ok

rmackay9, 2017-05-31:
It seems the "EKF: ok" message is always displayed even if the EKF isn't healthy.  To reproduce the problem:

- attach an oled display to a flight controller which has an external compass
- rotate the external compass until the ekf goes red
- note the "EKF: ok" remains

My guess is that only the case of an EKF failsafe does the EKF go bad.  But we currently don't allow an EKF failsafe to trigger while we're disarmed so this will never be seen.

This issue was raised on AC3.5 beta testing thread and confirmed by me (randy): http://discuss.ardupilot.org/t/oled-display-not-update-ekf/17954/4

oled wiki page: http://ardupilot.org/copter/docs/common-display-onboard.html

peterbarker, 2017-07-03:
This is designed to address the problem: https://github.com/ArduPilot/ardupilot/pull/6392

IamPete1, 2021-08-30:
Is this still a issue?

rmackay9, 2021-08-31:
@IamPete1,

I've retested and this issue still persists I'm afraid.