ArduPilot/ardupilot #6354: Custom Development for PixHawk

iskymaster, 2017-06-01:
#### Issue details
I apologize in advance for posting here but I am bit lost so need some help.

I am looking for a custom software developer for PixHawk for hire.

Any suggestions where to post such a thread and how to find one or two programmers?

#### Version
_What version was the issue encountered with_

#### Platform
[X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
QUAD's

#### Hardware type
PixHawk

#### Logs
_Please provide a link to any relevant logs that show the issue_

OXINARF, 2017-06-01:
We have a commercial support page (http://ardupilot.org/copter/docs/common-commercial-support.html) and you can also post in our forum (http://discuss.ardupilot.org/c/general/wanted-jobs).

closed 2017-06-01 by OXINARF

iskymaster, 2017-06-03:
@OXINARF - Thank you.