ArduPilot/ardupilot #6355: More Serial Ports support on PixHawk

iskymaster, 2017-06-01:
#### Issue details
I am running into a situation where I need more serial ports on PixHawk than whats available.

Are there plans to provide additional serial ports or are there any external multiplexors that i can use?

Thanks

#### Version
PixHawk 2.4.8

#### Platform
[X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
QUAD's

#### Hardware type
PixHawk

#### Logs
_Please provide a link to any relevant logs that show the issue_

magicrub, 2017-06-01:
No. Have you considered CAN? What do you have booked up right now?

closed 2017-06-01 by magicrub

iskymaster, 2017-06-01:
I need to connect the following devices to the PixHawk

1. GPS -1
2. GPS - 2
3. OSD
4. UAvionix ADS-B
5. Telemetry

and I might have one more device. Any suggestions?

jmachuca77, 2017-06-01:
OSD can share ports with telemetry. And I think you can use the console serial port also if you configure it as such.

iskymaster, 2017-06-03:
@jmachuca77 - You mean connect the OSD in parallel rather in and out? Please explain.

What do you mean by console serial port? Which port are you referring to as none are marked as console serial port.

Where can I find documentation.