ArduPilot/ardupilot #6362: Disabling GPS checks seems to force me to disable '16' aka 'INS(INertial Sensors - accels & gyros)

marcmerlin, 2017-06-01:
This report is valid as of AP 3.8rc5

As part of debugging on https://github.com/ArduPilot/ardupilot/issues/6261
I had many issues with likely a crappy GPS which prevented arming when I upgraded to AP 3.8rc5 (from 3.7.1).
[MAV 001:1] FAILURE: Failed CMD: 400
[MAV 001:1] PreArm: GPS HDOP 2.8 (needs 2.5)
[MAV 001:1] PreArm: GPS vert error 12.2m (needs
[MAV 001:1] FAILURE: Failed CMD: 400
[MAV 001:1] PreArm: GPS speed error 2.2 (needs 1.5)

Since I'm ok flying that plane with a no so good GPS (it's a small plane and I pretty much only hand fly it anyway), I wanted to override that check.
http://ardupilot.org/plane/docs/parameters.html#arming-check-arm-checks-to-peform-bitmask
hints that I need to disable '8' aka GPS Lock
However, disabling 8 from the arming bitmask did not allow arming.

Instead I had to disable '16' aka 'INS(INertial Sensors - accels & gyros)' and not '8' aka GPS Lock.
In other words, I have arming require 2+4+8+32+64+128+256+512 = 1006
This does not fill me with joy because I don't want to disable the check on gyros and accelerometers, but it seems that to disable the GPS check, I'm forced to.

This is honestly not very safe because having a not good GPS forces me to disable more important tests regarding gyros and accelerometers which are crucial for leveling and even FBWA, making my plane kind of unsafe to fly now since I've bypassed that check because of the GPS.
Could disabling '8' disable the prearm failures shown above?

marcmerlin, 2017-06-01:
@magicrub  Please remove the safety issue if I'm wrong about my assertion on safety.

OXINARF, 2017-06-01:
I've removed the Safety label because by default all checks are enabled. Disabling any check (and the problems that arise from it) is on the responsibility of the user.

The issue here is that the check of ARHS health is on the INS checks, but the AHRS health is dependent on EKF and EKF failure reason will 99% of the time be related to GPS not being good enough. This problem doesn't affect Copter (that has the AHRS check in GPS checks), but affects Plane and Sub - Rover apparently doesn't check the AHRS health but should.

A workaround for now is to change EKF settings to what GPS checks should be performed (http://ardupilot.org/plane/docs/parameters.html#ek2-gps-check-gps-preflight-check) or increase the checks threshold (http://ardupilot.org/plane/docs/parameters.html#ek2-check-scale-gps-accuracy-check-scaler).

marcmerlin, 2017-06-02:
Thank you @OXINARF, I will keep EK2_GPS_CHECK in mind  next time I have this issue.
But in the meantime, I think you agreed that on plane, disabling 8 from the CHECK matrix should disable all GPS related checks (in so far as they don't end up making INS checks fail enough that INS fails too).
Correct?

OXINARF, 2017-06-02:
Yes, I agree that the AHRS health check would be better in GPS checks than in INS checks. It isn't a black and white situation though because the AHRS health isn't necessarily about GPS.

marcmerlin, 2017-06-02:
Thanks. Mind you, if some end up disagreeing in the end, updating the documentation.  May be enough.
Right now I get a GPS error, I go to
http://ardupilot.org/plane/docs/parameters.html#arming-check-arm-checks-to-peform-bitmask
and I disable the GPS bit, but it doesn't work.

Would you like me to edit that to add 'GPS' to the INS column?
Also the columns were messed up and not aligned between the bitfield and the values, making it confusing.

So, I was just going to send you a CL, but the 'edit on github' button takes me to
https://github.com/ArduPilot/ardupilot_wiki/blob/master/plane/source/docs/parameters.rst
which is 404
Mmh, ok, I tried this:
saruman:~/RC/ardupilot_wiki$ find |grep parameters
./plane/source/docs/quadplane-parameters.rst
Where is the source of http://ardupilot.org/plane/docs/parameters.html#arming-check-arm-checks-to-peform-bitmask hiding?
In the process, I/we can also fix s/peform/perform/

magicrub, 2017-06-02:
The parameter page is special, it is generated by the source. You'll want to edit the ARMING library param CHECK [here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Arming/AP_Arming.cpp#L42).

OXINARF, 2017-06-03:
We don't want to edit that because INS checks are for inertial sensors, writing GPS there is misleading for Copter. I think this needs to fixed in code, not in documentation.

marcmerlin, 2017-06-03:
@OXINARF if you can get the code fixed in time for 3.8, I'll leave the docs alone :)

marcmerlin, 2017-06-06:
@OXINARF ok, I'm going to leave this bug with you.
If the code isn't going to be fixed in time for 3.8, assign it to me, and I'll fix the documentation to match reality.

tridge, 2017-06-07:
I agree with @OXINARF that it isn't a safety issue, but I also agree that flying without a GPS in a plane is not convenient now. That would be a nice improvement to have, but isn't a high priority for me.
You may find it best just to increase EK2_CHECK_SCALE, as suggested by @OXINARF, which will allow it to accept a worse GPS

amilcarlucas, 2017-10-22:
There are a couple of PRs from @peterbarker that increase the usage of AP_arming library and probably fix this. It would be nice if they get merged. @peterbarker does your PR fix this ?

marcmerlin, 2017-12-13:
ping @peterbarker , not sure if you saw @amilcarlucas' question

IamPete1, 2021-09-12:
I'm fairly sure this is fixed.

closed 2021-09-12 by IamPete1