ArduPilot/ardupilot #6363: AP 3.8rc5 does not flash without my holding the safety switch

marcmerlin, 2017-06-01:
This was originally reported in https://github.com/ArduPilot/ardupilot/issues/6261
I had a pixhawk and a pixfalcon. 
Both refused to flash 3.8rc5 to completion and gave me the first 3 notes of encounter of a 3rd kind, in a loop.
@peterbarker gave me the trick to hold the safety switch at boot to tell the firmware to flash the secondary CPU some other way, and that worked. 
After that, 3.8rc5 flashed ok.
However
1) one of my aircraft didn't even have a safety switch I could access, so this was a pain (I use arming by rudder)
2) if the primary way of flashing does not work, asking the user to use an esoteric secondary way is probably not necessary and it should just default to the secondary/working way.

marcmerlin, 2017-06-06:
I flashed my last aircraft, also a genuine pixhawk, and same thing, the flash failed unless I held the safety switch at boot.
So that's 3 failures out of 3 on two different kinds of pixhawk-like FC
I'm going to assume it impacts all pixhawks.

WickedShell, 2017-06-06:
I haven't been having this problem (nor have any of my direct user base), except on one Pixhawk which is several years old. I'm guessing this is related to what version was present. I'm not sure how wide spread this will be across the board though for everyone. (I've only tested on 3DR PH, PH2 and hex Cubes...

magicrub, 2017-06-06:
Marc, do you have a very very old pixhawk by any chance? Just wondering.
Trying to figure out why it's not working for you and does for everyone
else..

WickedShell, 2017-06-06:
Was discussed on the dev call, happens to relate to a bug with PX4IO firmware that was fixed in master that was inhibiting the FMU from getting the IO processor back into the bootloader. It should be a one time problem, but we are looking into either a intermediate 3.7.2 that resolves the upgrade path for users who don't have the button or to see if we can find a better fix to not require all users to dig out buttons.

marcmerlin, 2017-06-06:
@magicrub the pixfalcon was bought last year, no idea when it was made.
The pixhawks were bought directly from 3DR 3 and 2 years ago.

magicrub, 2017-06-06:
That's considered new. Thx.

Naterater, 2019-01-03:
Is this still relevant after switching to ChibiOS IO firmware?

WickedShell, 2019-06-27:
Would be relevant as it was a IO firmware problem, but we are now a full stable version on that has fixed the problem, so clearly we aren't going back to make an intermediate firmware that would make the upgrade work better.

marcmerlin, 2019-07-12:
how about timing, I just hit this problem again.
pixhawk upgraded to 3.9.x. It had a bug I couldn't fly with, flashed 3.8.5 and same problem again.

marcmerlin, 2019-07-12:
I had to follow this procedure again, and then it booted.
https://github.com/ArduPilot/ardupilot_wiki/blob/master/dev/source/docs/pixhawk-force-px4io-flash.rst

marcmerlin, 2019-07-12:
generally it would be super cool if 
https://docs.px4.io/v1.9.0/en/getting_started/tunes.html
actually had the 3 tones in there, with a pointer to the force flash doc

auturgy, 2019-07-12:
Perhaps http://ardupilot.org/plane/docs/common-sounds-pixhawkpx4.html ?
The px4 docs aren’t really relevant for ArduPilot.

Regards,

James

marcmerlin, 2019-07-13:
Thanks @auturgy , the page you point to is much better, but indeed not what google pointed me to.
Now, on the page you give, the tunes don't match either
http://ardupilot.org/plane/docs/common-sounds-pixhawkpx4.html
http://download.ardupilot.org/downloads/wiki/pixhawk_sound_files/PX4_ReadyToUploadIOBoardFirmware.wav
is close to what I hear, but I only hear 3 tones

http://download.ardupilot.org/downloads/wiki/pixhawk_sound_files/PX4_CompletedIOBoardFirmwareUpload.wav
is actually what I heard but without the 2 beep beep at the end of that wav, just the loop of the first 3 notes, forever

But putting the not quite matching sound aside, this original bug was that I couldn't flash a version of AP without holding the safety switch and without getting into a state that honestly wasn't obvious and made me lose 1.5h on the field and a microsd card after trying to eject it thinking I was getting a bad format error (the beep code thing honestly needs a rework, bios beeps of short and long are much simpler to look up than music tones with notes that you have to match with wav files (and none of them really matched actually)).
So, 2 years later, the problem is still there, downgrading from 3.9.8 to 3.8.5 got me in that non booting situation.
Is there a way to make it not happen in the first place?  (@magicrub  this was the issue I had this afternoon BTW).

marcmerlin, 2019-07-13:
and actually if it is "normal" to have to hold the safety switch for some kinds of upgrades, then mission planner should tell you so. Instead it told me to wait for the music to stop playing, which it never did, but I was worried about unplugging because maybe I would power down the unit during a critical phase of the flash and maybe brick it.

marcmerlin, 2019-07-13:
I did more flashing to older releases and every time I had to try several times and play with the safety switch for it to eventually work.
I didn't downgrade to anything below 3.8.0 (and first time was a downgrade to 3.8.5).
@WickedShell  mentioned it was fixed around 2017. Which actual AP release is that fix in? (just to know if 3.8.5 is before that fix or after)

auturgy, 2019-07-13:
You shouldn’t have to hold the button for any upgrades anymore: the IO firmware should update automatically.
One thing that might be worth trying is updating the bootloader, which is a lot easier now than it used to be.
You can do that via Missionplanner “CTRL-F” screen http://ardupilot.org/dev/docs/bootloader.html

marcmerlin, 2019-07-13:
Got it, so the bootloader does not get updated along with the firmware upgrade?

auturgy, 2019-07-13:
It’s packaged as payload within the firmware (so is physically on the device) but doesn’t overwrite the existing bootloader automatically. You have to command the upgrade.

Regards,

James

marcmerlin, 2019-07-13:
gotcha, so given that I'm getting failures from MP when trying so, are there instructions on how to do it, whether I have to get in some special mode for the command to be accepted, or something?
I tried while connected in mavlink, while disconnected, and when rebooting. All 3 options failed.

IamPete1, 2021-04-12:
I'm going to close this, if it  is still at issue on 4.0 feel free to re-open this or make a new issue.

closed 2021-04-12 by IamPete1