ArduPilot/ardupilot #6366: EKF3 has an error in NavEKF3_core::setWindMagStateLearningMode()

luweikxy, 2017-06-02:
#### Issue details
in the last lines of function NavEKF3_core::setWindMagStateLearningMode(), showned as follow:

    // Adjust the indexing limits used to address the covariance, states and other EKF arrays to avoid unnecessary operations
    // if we are not using those states
    if (inhibitMagStates && inhibitWindStates && inhibitDelVelBiasStates) {
        stateIndexLim = 12;
    } else if (inhibitMagStates && !inhibitWindStates) {
        stateIndexLim = 15;
    } else if (inhibitWindStates) {
        stateIndexLim = 21;
    } else {
        stateIndexLim = 23;
    }

there is an error in " } else if (inhibitMagStates && !inhibitWindStates) { "
"!" in this line should be deleted.

#### Version
"APM:Copter V3.6-dev"
or
"ArduPlane V3.8.0beta5"

rmackay9, 2017-06-02:
@luweikxy,
Thanks for the contribution, i have ping'd paul so i hope he can have a look.  We hear there are many users in china, its really great to hear from you!

luweikxy, 2017-06-02:
@rmackay9 
i am very glad to get your reply, you said "We hear there are many users in china, its really great to hear from you!", so it gives me a sense that you are not familiar with chinese apm uesrs, but... i know you have been invited in a chinese apm WeChat group named "Indoor Loiter" and you sometimes interact with the members in it including me, so it is obviously that you very familiar with chinese apm uesrs, dont pretend, hahaha...

priseborough, 2017-06-02:
Thank you for your observation. This piece of code has been reworked by this PR https://github.com/ArduPilot/ardupilot/pull/6288

Let me know what you think.

luweikxy, 2017-06-02:
@priseborough 
how i find this error: i have read your ekf(1) and ekf2, so i am relatively familiar with ekf, then i began to read ekf3, and when i read this line where this error occurs, it is easy to find this error only needs you are careful.
only when inhibitMagStates is true and inhibitWindStates is true, then the stateIndexLim can be 15. if  inhibitMagStates is true and ! inhibitWindStates is true, then it means that WindStates are not inhibit, so the stateIndexLim must be 23.

by the way, there is a piece of code in EKF2 has puzzled me since i first saw it. that is FuseDeclination(0.34f) in EKF2,  showed as follow:
    // Calculate the innovation
    float innovation = atanf(t4) - magDecAng;
why the predicted value is atanf(t4)? isnt that should be atan(magE , magN)?? why not use atan(magE , magN)?? this really really has confused me from then to now, let me cry for a while....
luckily, the same peice of code in EKF3 has corrected to:
    // Calculate the innovation
    float innovation = atan2f(magE , magN) - magDecAng;
but i still want to know how to understand "float innovation = atanf(t4) - magDecAng;" in EKF2,  what did you think when you writed atanf(t4) but not atan2f(magE , magN) in EKF2?  i would be very appreciate it very much if you explain this when available.

ranqingfa, 2017-06-02:
Great work! LuWei

priseborough, 2017-06-03:
Good catch @luweikxy. The temporary t4 in a earlier version of the auto generated code was =  magE/magN. Unfortunately when the auto generated code was updated, the innovation calculation was not updated to match. I will submit a fix.

priseborough, 2017-06-03:
https://github.com/ArduPilot/ardupilot/pull/6374

luweikxy, 2017-06-03:
@priseborough 
I am a little confused about what you said. I have read your matlab code for c code generation before i began to read EKF in apm, that is easy to understand and i think it is totally ok, as follow:
    // Calculate observation Jacobian and Kalman gains
    // angMeas = atan(magE/magN);
    // H_MAGD = jacobian(angMeas,stateVector); % measurement Jacobian

But... in apm c++ code of EKF2, the corresponding code is:
    float t2 = magE*magE;
    float t3 = magN*magN;
    float t4 = t2+t3;
    ...
    float innovation = atanf(t4) - magDecAng;
that is to say: here t4 = t2+t3 = magE*magE + magN*magN;
here is the question, which totally confused me for almost three months: i konw t4 must be atan(magE/magN) essentially, but how can (magE*magE + magN*magN) equals to (magE/magN)? I want to cry...May be (magE*magE + magN*magN) is the taylor expansion of (magE/magN)? no no no...obviously not. I have asked several person, none of them can give me an answer. they are just as confused as me, one of them is a professor named GongMin Yan in School of Astronauticsin in North Western Polytechnic University in China( I am a postgraduate from NWPU in School of Aeronautics and have graduated for two years, my first job is in a national UAV research institute for both military and civil use, but i think you also kown that the salary is too low and the atmosphere is dull and management is chaotic in this kind of national institute because i have seen your resume that you had been working in BAE for a long time and i cant see any hope, so I jumped to a uav start-up company for industry use about half yaers ago, and I want to go to doctor to study machine learning after one or two years.   ok let's back to the question...I say too much...)

Why I am a little confused about what you said? because you said "The temporary t4 in a earlier version of the auto generated code was = magE/magN.", but t4 in apm actually equals to t2+t3 = magE*magE + magN*magN. Then you said "Unfortunately when the auto generated code was updated, the innovation calculation was not updated to match. ", Here your "not to match"  translates in my mind to that it  is right but just not be updated , but i think it is wrong, because:
    t4 = magE*magE + magN*magN     DO NOT equals to        magE/magN
So, I want to confirm that "innovation = atanf(t4) - magDecAng;" in EKF2 is totally wrong, or it is indeed right but just not be updated or not match? And if it is right, how to understand t4 = magE*magE + magN*magN in apm C++ code?

OXINARF, 2017-06-03:
@luweikxy Paul has already fixed the problem in PR #6374, which I have already merged. What Paul explained was that `t4` is a temporary variable auto-generated from Matlab code, which in a previous version generated `t4 = magE/magN`, so having `float innovation = atanf(t4) - magDecAng;` was correct. When Matlab code was updated and `t4` changed to being generated as `t4 = t2+t3;`, the `innovation` variable should have been updated to use `magE/magN` instead of using `t4`, but wasn't.

So it was good catch from you! Next time you think you have found a bug, tell us in our forum (http://discuss.ardupilot.org) or in Gitter chat (http://gitter.im/ArduPilot/ardupilot).

luweikxy, 2017-06-04:
@OXINARF 
Many many thanks! Your explain is very clear, I've got it.
As this is the first time to try to submit bug, next time I will go to the two websites you give me.

luweikxy, 2017-06-06:
@priseborough 
Hello, I am sorry to trouble you agin and agin, I didn't mean to...
I find a small problem in EKF2/3 in function NavEKF3_core::alignMagStateDeclination(), this function calls function NavEKF3_core::FuseDeclination(float declErr), the question is that FuseDeclination() here seems no useful. Because that in alignMagStateDeclination(), stateStruct.earth_magfield has already been set to:

stateStruct.earth_magfield.x = magLengthNE cosf(magDecAng);
stateStruct.earth_magfield.y = magLengthNE sinf(magDecAng);

So in FuseDeclination(), this peice os code eques to zero:

// Calculate the innovation
float innovation = atan2f(magE , magN) - magDecAng;

Why innovation is zero? Because here:

float magN = stateStruct.earth_magfield.x; = magLengthNE cosf(magDecAng);
float magE = stateStruct.earth_magfield.y; = magLengthNE sinf(magDecAng);

So atan2f(magE , magN) = magDecAng! Because both magDecAng in the two function are set by 

    float magDecAng = use_compass() ? _ahrs->get_compass()->get_declination() : 0;

, so the two magDecAng are supposed to the same. So the innovation is zero, so FuseDeclination() called here seems of no ues, the only use I think is to reduce the P (error between estimated value and true value).

priseborough, 2017-06-06:
Please tell us in our forum (http://discuss.ardupilot.org) or in Gitter chat (http://gitter.im/ArduPilot/ardupilot).

closed 2017-06-06 by priseborough