ArduPilot/ardupilot #6370: u-blox 1 HW: 00080000 SW: 2.01 (75350)  can be mis-detected as NMEA on 3.8rc5

marcmerlin, 2017-06-02:
I'm splitting this from https://github.com/ArduPilot/ardupilot/issues/6261 to keep separate issues separate.
@WickedShell when I did more tests on 3.8rc5, my GPS sometimes worked although I had arming issues that seemed to be due to genuine jamming from a mobius camera now causing big RF noise that it wasn't before (I need to debug this separately).
however, just right now, I had a boot with 
[MAV 001:1] GPS 1: detected as NMEA at 38400 baud 
[MAV 001:1] GPS 1: detected as NMEA at 115200 baud 
[MAV 001:1] GPS 1: detected as NMEA at 38400 baud 
[MAV 001:1] GPS 1: detected as NMEA at 115200 baud 
[MAV 001:1] GPS 1: detected as NMEA at 38400 baud 
[MAV 001:1] GPS 1: detected as NMEA at 115200 baud 
[MAV 001:1] GPS 1: detected as NMEA at 38400 baud 
[MAV 001:1] GPS 1: detected as NMEA at 115200 baud 
logfile: 17-06-01_23-31-26.bin

and then the next boot with
[MAV 001:1] u-blox 1 HW: 00080000 SW: 2.01 (75350) 
[MAV 001:1] Airspeed sensor calibrated 
[MAV 001:1] GPS 1: detected as u-blox at 115200 baud 
logfile: 17-06-01_23-37-12.bin

I did nothing outside of rebooting between the 2.
I *think* I didn't have this problem with 3.7.1, but considering that I've been messing with cabling a lot since then, it's not a 100% guarantee.
Also, on the plus side, most boots seem to get the GPS working now.

marcmerlin, 2017-06-02:
[dataflash.zip](https://github.com/ArduPilot/ardupilot/files/1048445/dataflash.zip)

WickedShell, 2017-06-02:
This actually can't happen anymore as master no longer will auto detect NMEA :) https://github.com/ArduPilot/ardupilot/pull/6346 (Was a change that I sent in 3 days ago :P ) To keep the issues list related to code issues lets move this to a discuss topic :) I'll go through the DF logs and open a post on discuss and tag you on it.

closed 2017-06-02 by WickedShell

marcmerlin, 2017-06-02:
Sounds good, thanks for both the fix in master and looking at the logs @WickedShell  (as well as the helpful replies as always)

marcmerlin, 2017-06-08:
To put some closure here, I've had multiple issues with this GPS later on. It was a Neo N8M and I was able to upgrade it to firmware 3.x (from 2.x). This seems to have helped with its stability and reliability.
I've also confirmed that when I connected it with make shift wiring and jumper wires, it was detected as NMEA, so it indeed seems that suboptimal wiring did cause this problem. Thanks again for removing that autodetection, it will hopefully help remove confusion for people with not so good wiring.