ArduPilot/ardupilot #6375: Plane: automatic air brakes support

palm369, 2017-06-03:
We need some code for an automatic control of airbrakes on plane. We already have code for using differential spoilers (manually) and auto flap deployment. But the flaps are just set by the demanded airspeed and not by actual speed. This means that the flaps are deployed even if the higher lift (Ca value) is not necessary because the plane is still fast.

I do distinguish between:
- Flaps: used to gain higher Ca and therefor lower stall speed
- Airbrakes: used to slow down the plane

In my case I fly a flying wing which does glide well and has only two control surfaces. It is kind of impossible to get good automatic landings without using a very low glide path angle (like <5 deg).
If the angle is steeper, the plane does not archive the desired landing speed and is too fast on touchdown (critical on some planes with not super rigid undercarriage).
So I could put airbrake on it which are actuated by an additional servo or two and use the "do set servo" command in a mission. Or I could use differential spoilers and set them manually somehow. Or use reverse thrust, which also not controlled by actual airspeed.
What is missing is an automatic option to use flaps or diff. spoilers or air brakes for this.

Benefits:
- loose speed / altitude faster if demanded
- allow steep landing approaches without overspeeding
- independent control of alt and spd possible -> more accurate landings
- TECS_LAND_SPDWGT an LAND_PF getting obsolete

How this could be implemented:
- Param for what to use for slowdown (Flaps, airbrakes, etc)
- Params for a PID controller, or just a P from speed-delta
- Param for min alt to use this, so airbrakes not getting damaged on landing
- Param for max speed to use this (so not getting damaged by high loads)
- Logic to prevent activation by wind gusts / only usable while sinking is demanded
- ...?

Would love to write the code myself, but I'm not a programmer...

magicrub, 2017-06-03:
What kind of flying wing do you fly and/or anticipate using with this?

magicrub, 2017-06-03:
Also, something you did not mention at all is reverse thrust. I land a Skywalker x8 fully loaded at 20 deg slope just fine.

marvinlange, 2017-06-03:
I would love this and be willing to try on my RV-Jet as it's glide slope very difcult on short landing.  lately when  doing areal mapping for people with small LZ I have litterly just stall it all the way into high grass

magicrub, 2017-06-03:
Btw, flaps are already implemented and are triggered by current airspeed. 

>Or use reverse thrust, which also not controlled by actual airspeed

I'm not sure where your getting your info. Reverse thrust is indeed using airspeed current and demand and is infact even better than that. It's deeply embedded into the tecs controller and is used in the pitch/thrust controller which controls airspeed. If you need to slow down, it will slow you down in a very controlled and delivered way.

palm369, 2017-06-04:
I use a HK SkyRay aka RMRC Mako, but I also had even more problems with that on a very high loaded RV-Jet

AFAIK flaps are just set by demanded airspeed, not actual (mentioned here: http://ardupilot.org/plane/docs/automatic-flaps.html)
Nothing aditional mentioned here: https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/release-notes.txt

Reverse thrust: Got my infos here and actual airspeed control is not mentioned: http://ardupilot.org/plane/docs/automatic-landing.html?highlight=reverse%20thrust#reverse-thrust-landing
I never tried that option. If its really working like you say, this would be great. Although the complexity of setting this up is quite high and a lot of people need to change and/or reflash their ESC for this. Also when you land with almost flat battery this not doing well.
IC engines can't use this option.

If this option is already implemented, it should be easy to output this on a servo additionally. So one could set a bitmask on what to use as output:
- rev thrust
- flaps
- airbrakes
- diff. Spoilers

geofrancis, 2017-09-28:
i was looking at using crow mixing for airbrakes but i see its not possible using the current firmware as you cant set more than one function to an output.

magicrub, 2018-03-08:
@palm369 I just checked. YES, the flaps are controlled by TARGET airspeed and not current airspeed. However, that works for slowing you down. If you're too fast, and you want to go slower and that is below the flap airspeed threshold (FLAP_1_SPEED), then they will deploy to (FLAP_1_PERCNT).

magicrub, 2018-03-08:
Here's reverse thrust of a Skywalker X8 at 45deg. That's a bit of an extreme and it bounces a bit on the hard ground, but it works great to keep the airspeed low.
https://www.youtube.com/watch?v=XFZfSSTQgI4

magicrub, 2018-03-08:
back to your original question: air brakes. I mentioned it in my talk recently.
https://youtu.be/XuSl9Io93aQ?t=579

Here's the part
https://hobbyking.com/en_us/glider-spoiler-servoless-left-and-right-side.html

The way I've seen them used is to simply manually engage them as part of your landing procedure which slows you down and then if you're too slow then throttle compensates. The root of your initial post was to make this an automatic feature which is a good idea. But my question is: How?

We don't want them to kick in if we're just 0.1m/s too fast.. so when do they engage? For this particular type they are 100% or 0%, nothing in the middle (or at least not meant to be). You coudl of course make something custom. But when do they engage? Only during landing? Any other time? I had the same question when I made reverse thrust so I added param [USE_REV_THRUST](http://ardupilot.org/plane/docs/parameters.html#use-rev-thrust-bitmask-for-when-to-allow-negative-reverse-thrust) to only allow it in certain circumstances. I think it's at least a good step to only allow it during landings and go from there.

Btw, this is easy to add into the source, but it is also something we can easily add as [a script](https://www.youtube.com/watch?v=nc8IwU2G1rY) in the future.

palm369, 2018-03-15:
HOW: Just use the same logic which also defines the amount of reverse thrust given and make it possible to output this to a separat channel
-> new servo function: "airbrake"
-> servos assigned with that will be steered via the rev. thrust logic

for the brakes which can only be not or fully deployed: Use threshold for speed difference with hysteresis
eg: aim: 15, deploy > 16, move in < 15

magicrub, 2020-09-17:
This has been implemented in a few different ways. We have airbrake flight surfaces and reverse thrust for normal flight, including landing patterns via https://github.com/ArduPilot/ardupilot/pull/15343

closed 2020-09-17 by magicrub