ArduPilot/ardupilot #6378: Copter: do_take_picture should be able to be triggered even without GPS

rmackay9, 2017-06-05:
This is an edge case but at the moment "do_take_picture" is triggered from the "Copter::update_GPS" but it's theoretically possible that we are using a different position estimate source beside the GPS (i.e. visual odometry).

OXINARF, 2017-06-05:
@rmackay9 I've updated the title as this affects all vehicles. Feel free to change it again if you don't find it clear enough.

7princekumar, 2018-10-07:
I would like to work on it but have no clue . I am new to open source contribution. I need your help to solve this.

WickedShell, 2020-07-17:
Done in #14804

rmackay9, 2020-07-17:
Except that this line still exists meaning the camera trigger still won't work without a GPS. https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Camera/AP_Camera.cpp#L344

WickedShell, 2020-07-18:
Hmm, yeah okay, I yield there, that should probably be addressed as well :)

closed 2024-08-06 by peterbarker