ArduPilot/ardupilot #6380: GLOBAL_POSITION_INT message sent when there is no GPS data available

DonLakeFlyer, 2017-06-05:
This was tested using latest ArduCopter RC candidate:

If you boot up a vehicle prior to any sort of GPS information being available ArduPilot will still send GLOBAL_POSITION_INT messages with incorrect 0/0 lat/lon values in it. I verified that no GPS information was available by seeing that the GPS_RAW_INT message was being sent and GPS_RAW_INT.fix_type= GPS_FIX_TYPE_NO_FIX.

This is makes it very confusing from a GCS standpoint to figure out whether the 0/0 lat/lon value is real or not since 0/0 is a valid location. I can certainly special case this for ArduPilot. But that sort of thing just seems odd and a bad thing.

What happens with PX4 Pro stack is if there is no valid GPS location then GLOBAL_POSITION_INT is not sent.

OXINARF, 2017-06-05:
Just quoting what I wrote in Gitter:

> Sending GLOBAL_POSITION_INT is part of a stream and the truth is that even if you don't have a GPS fix it doesn't mean we don't know our global position. For example, let's say you took off with GPS, but at some point it broke and you lost GPS; if you have optical flow, it should be good enough to continue having our global position - I'm almost sure that EKF will continue to provide global position information in that case. Also, our beacon feature requires that you give initial global position and from that point forward, provided the beacons are working properly, you have global position information but no fix. Like @khancyr said, for now, checking 0/0 should be good as our code assumes in a lot of places that's not a valid position. I believe there can be a way we can stop sending the message if we don't have any global position information though

rmackay9, 2017-06-05:
I think we can generalise this so that if we don't have a position estimate, we don't send it.

WickedShell, 2017-06-05:
I'd like that this stream continued regardless of GPS or not... It's hard to assess if this message is even present...

rmackay9, 2017-06-05:
@WickedShell, so your suggestion is to leave it as-is and the GCS should interpret 0,0,0 as invalid?

WickedShell, 2017-06-06:
That's definitely my preferred approach.

DonLakeFlyer, 2017-06-06:
Seems pretty strange to me. If you are just going to ignore it when send it?

WickedShell, 2017-06-06:
I'm not actually ignoring it. If I'm not getting GLOBAL_POSITION_INT messages to the GCS I consider that an indication that something is wrong with the stream rates and start moving to fix that (on the assumption that I missed a param change or similar). I'd rather see the GCS have the option of ignoring it if desired. Just my $0.02

DonLakeFlyer, 2017-06-06:
Ah, get it now. Not changeable then.

WickedShell, 2017-06-06:
I mean we can decide my implementation is wrong, I'm definitely relying on undefined behavior :)

DonLakeFlyer, 2017-06-06:
Hmm. Are you using stream sets? Could you trigger off of a different message from the set?

amilcarlucas, 2017-10-22:
I think that sending the message is a good thing. One can detect that the mavlink communication is up, stream rates are correct, but the global position has not been found or is lost.
0,0,0 is a very improbable location, don't you think ? maybe a better fix would be to use 91,181,65535 but that (invalid) one also needs special treatment in the GCS

WickedShell, 2018-09-02:
Close this out. `GLOBAL_POSITION_INT` carries a number of fields that we may know without GPS fix, and are still feasible to extract from the message. This is particularly valuable as it allows a GCS to grab more information from a single packet (when GPS is present) and doesn't make the fallback case(s) any worse without GPS.

closed 2018-09-02 by WickedShell

ardugitter, 2021-12-27:
Just wondering why this was closed.

The only useful field filled in by this message when there is no GPS lock is the heading.

If you want the heading, you can use MAVLINK_MSG_ID_ATTITUDE

It is dangerous and confusing to send the GLOBAL_POSITION_INT message when the position is not known in my opinion.

Please could we re-open this and stop this message being sent unless there is a GPS lock? Thanks.

peterbarker, 2021-12-27:
On Mon, 27 Dec 2021, ardugitter wrote:

> The only useful field filled in by this message when there is no GPS lock is the heading.

Global position can also be sourced from an origin set from a GCS and a 
visual navigation system.  Amonst other things.

ardugitter, 2021-12-28:
Okay, I'll just have to make sure I don't fly near lat 0, lon 0 in future. :-)

MoBaT, 2022-06-28:
I'm also not a fan of it outputting 0,0,0 when GLOBAL_POSITION_INT is not known or has not been set through an actual GPS or an origin set.