ArduPilot/ardupilot #6382: Should AUTO activate if certain conditions are not met? (like gyros not calibrated)

marcmerlin, 2017-06-06:
AP 3.8rc5.
So, I'll start by saying that ultimately the plane crash is my fault and responsibility, however AP could potentially have refused to activate auto if something critical like gyros weren't working.

What happened:
1) my bixler that had repeated problems with the cameras making the GPS signal marginal and prevented arming
2) when on the field, I used tower, and sadly due to https://github.com/DroidPlanner/Tower/issues/1865 it didn't give me any messages from mavlink, so it left me in the dark about the real reason.
Sadly, I stupidly guessed that arming was for the 20th time, a GPS issue, and overrode it (I did not need GPS for the flight I was going to make)

Yes, I'm upset that tower hid critical information I needed to have, and will look for another android ground station that does not do this (should there be one). Andropilot does the right thing, but 1) seems like a dead project sadly, 2) does not seem to plot my aircraft route on the screen, which tower does do.

I then activated auto and although the gyros were not working (didn't know this of course), auto did activate, and promptly took off, tipped over to the side, and crashed at full power into the ground, being unable to level the plane.

Should auto refuse to activate if some critical things like gyro aren't working? 

Messages:
2017-06-05 13:05:30.66: MSG {TimeUS : 39104148, Message : Airspeed calibration started}
2017-06-05 13:05:30.66: MSG {TimeUS : 39104178, Message : Ground start complete}
2017-06-05 13:05:30.98: MSG {TimeUS : 39421270, Message : ArduPlane V3.8.0beta5 (708b483d)}
2017-06-05 13:05:30.98: MSG {TimeUS : 39421286, Message : PX4: 8d505a02 NuttX: 1a99ba58}
2017-06-05 13:05:30.98: MSG {TimeUS : 39421334, Message : PX4v2 00380039 33355114 35303335}
2017-06-05 13:05:30.98: MSG {TimeUS : 39421350, Message : New mission}
2017-06-05 13:05:32.35: MSG {TimeUS : 40787687, Message : Airspeed sensor calibrated}
2017-06-05 13:05:33.13: MSG {TimeUS : 41575777, Message : GPS 1: detected as u-blox at 115200 baud}
2017-06-05 13:05:33.87: MSG {TimeUS : 42315956, Message : EKF2 IMU0 initial yaw alignment complete}
2017-06-05 13:05:33.88: MSG {TimeUS : 42316970, Message : EKF2 IMU1 initial yaw alignment complete}
2017-06-05 13:05:34.38: MSG {TimeUS : 42818111, Message : Init HOME}
2017-06-05 13:05:38.99: MSG {TimeUS : 47435943, Message : EKF2 IMU1 tilt alignment complete}
2017-06-05 13:05:39.17: MSG {TimeUS : 47615289, Message : EKF2 IMU0 tilt alignment complete}
2017-06-05 13:05:45.20: MSG {TimeUS : 53639257, Message : u-blox 1 HW: 00080000 SW: 2.01 (75350)}
2017-06-05 13:06:27.82: MSG {TimeUS : 96257548, Message : EKF2 IMU0 Origin set to GPS}
2017-06-05 13:06:28.02: MSG {TimeUS : 96458061, Message : EKF2 IMU1 Origin set to GPS}
2017-06-05 13:06:28.64: MSG {TimeUS : 97077087, Message : EKF2 IMU0 is using GPS}
2017-06-05 13:06:28.84: MSG {TimeUS : 97277573, Message : EKF2 IMU1 is using GPS}
2017-06-05 13:09:10.40: MSG {TimeUS : 258839060, Message : PreArm: Gyros not calibrated}
2017-06-05 13:09:13.44: MSG {TimeUS : 261878194, Message : PreArm: Gyros not calibrated}
2017-06-05 13:10:05.60: MSG {TimeUS : 314039058, Message : PreArm: Gyros not calibrated}
2017-06-05 13:10:14.66: MSG {TimeUS : 323098661, Message : PreArm: Gyros not calibrated}
2017-06-05 13:10:22.22: MSG {TimeUS : 330658591, Message : PreArm: Gyros not calibrated}
2017-06-05 13:10:25.26: MSG {TimeUS : 333698306, Message : PreArm: Gyros not calibrated}
2017-06-05 13:10:28.28: MSG {TimeUS : 336718396, Message : PreArm: Gyros not calibrated}
2017-06-05 13:11:24.84: MSG {TimeUS : 393278168, Message : Throttle armed}
2017-06-05 13:13:09.44: MSG {TimeUS : 497881510, Message : Executing nav command ID #22}
2017-06-05 13:13:09.44: MSG {TimeUS : 497881595, Message : Resetting previous waypoint}
2017-06-05 13:14:01.88: MSG {TimeUS : 550319173, Message : Armed AUTO, xaccel = 6.7 m/s/s, waiting 0.0 sec}
2017-06-05 13:14:01.88: MSG {TimeUS : 550319276, Message : Triggered AUTO. GPS speed = 0.4}
2017-06-05 13:14:06.00: MSG {TimeUS : 554442443, Message : Holding course 2886 at 13.6m/s (32.4)}

It may be that the answer is "no", if so feel free to close this.
[00000012.BIN.gz](https://github.com/ArduPilot/ardupilot/files/1053814/00000012.BIN.gz)

WickedShell, 2017-06-06:
Given that we just had an extensive discussion about allowing modes that need RC input without requiring RC be present before I'm inclined to say the answer is auto should still be allowed. Among other use cases this is an option that developers use at the moment while testing.

As you acknowledged disabling the arming checks places the risks solely in your hands. If you were convinced it was GPS errors then I would recommend only unsetting the relevant bits rather then just force disabling the entire check set.

Out of curiosity does tower not make status text messages available anywhere?

marcmerlin, 2017-06-06:
@WickedShell I agree about disabling specific bits, but tower makes it not really possible unless you brought the bitfield documentation with you on the field, and you do binary math. The end user interface is very lacking in this case (on top of the fact that you have to disable INS instead of GPS as per https://github.com/ArduPilot/ardupilot/issues/6362, so by disabling INS I'd have disabled the gyro test and armed anyway)
And no, I don't think tower has as way to give me mavlink messages and speak them out, like andropilot does (at least it's not default and if there is an option, I didn't find it).
But back to auto without gyro. Clearly, this cannot work, can it?
Is there really a good reason to allow it outside of my likely being the first person to crash or report that specific case?

WickedShell, 2017-06-06:
Auto without calibrated gyro not well, this is definetly true, but it either might not be to far off, or at least the use case I'm thinking of is that I've used this to test servo outputs/camera triggers on the bench before.

@magicrub @tridge any opinions either way?

magicrub, 2017-06-06:
To rephrase the scenario: if we are indoors without GPS you are requesting we should inhibit the switch to any mode that needs GPS such as auto? Even if you remove the safeguards which are debating enabled so it's absolutely impossible to switch to about half of the available modes while indoors? 

Copter does this, all mode changes get checked and can be rejected if certain criteria is not met. Plane has chosen not to do that because be in many cases the aircraft is flying away from you and if you think it's in over mode when actually in another you have a much higher chance of losing the aircraft than you would vs copter.

However, this may be a worthwhile thing to make a Param: mode_change_always_successful.

peterbarker, 2017-06-06:
On Mon, 5 Jun 2017, Marc MERLIN wrote:

> So, I'll start by saying that ultimately the plane crash is my fault and
> responsibility, however AP could potentially have refused to activate auto
> if something critical like gyros weren't working.

Sorry, but this is exactly what the arming checks are for.  Defeating the 
checks is going to cause this (and other problems).  "More than enough 
rope", as they say.

Two suggestions:
  - try QGroundControl on your Android device; the developer is keen to 
find and fix problems with ArduPilot support!
  - have another fully-fledged GCS handy that you can pull out in dire need

At the moment nobody has really considered hanging the Notify LCD screens 
on planes AFAIK - but they might have given you sufficient information to 
ward you off arming in this instance!

Yours,
-- 
Peter Barker                          |   Programmer,Sysadmin,Geek.
pbarker@barker.dropbear.id.au	      |   You need a bigger hammer.
:: It's a hack! Expect underscores! - Nigel Williams

marcmerlin, 2017-06-06:
@peterbarker I totally understand, I did state in my first comment that if it doesn't sound reasonable, it's fine to close. I'm leaving this up to you :)

magicrub, 2017-06-06:
I have some of those LCDs, would you like to try one out?

closed 2017-06-06 by magicrub

marcmerlin, 2017-06-06:
@magicrub thanks for the offer. It could be cool to look at but ultimately the problem is with the ground station. Its job is to tell me pertinent information and it utterly failed at doing so.
I'll try qgroundcontrol to see if it does a better job, but I really hope tower gets fixed given that it's likely what most people are using.
Speaking of which, night_ghost minimosd also has a filter to remove mavlink messages before arming it seems (I checked my goggles and didn't get them, but I do get mavlink after aircraft is armed). That should be easy to rip out, so I'll give that a shot in a little while.

magicrub, 2017-06-06:
I think it's Bill who is maintaining Tower now, right?

marcmerlin, 2017-06-06:
Yes, it seems to be, and last I talked to him, he was quite busy and with little time to spend on apmplanner, so I'm assuming it's the same problem with Tower. I can't blame him, I know how those things go.
Either way, I'll give qgroundcontrol a try after I figure out if I can glue/solder this plane back into working shape.

marcmerlin, 2017-06-06:
@magicrub just got qgroundcontrol working on my phone now (never could get it to connect to my USB serial adapter before). It gives all the mavlink messages before arming, so this is actually what I needed from tower.
I haven't flown with it yet, but it looks like it'll work fine, thank you.
Strangely with all the options it has, it seems to be missing the "reset airspeed sensor" that mission planner has, but oh well, it's not as critical.

WickedShell, 2017-06-06:
@DonLakeFlyer is there a way on QGC on mobile to preform an airspeed calibration?

marcmerlin, 2017-06-10:
@WickedShell I opened https://github.com/mavlink/qgroundcontrol/issues/5274 and it'll be added to a future version