ArduPilot/ardupilot #6383: How to set Waypoints in RAM (volatile)  just like PX4 Firmware??????

RQRobert, 2017-06-06:
#### Issue details
_Please describe the problem, or desired feature_
I am having trouble in AUTO mode, 
When i trigger AUTO ,  i forget to update waypoints
and my quad flies to the previous waypoints (location) .
I have 2 crashes due to this and i feel PX4 firmware has solution to this problem.
But PX4 firmware is not as stable as Arducopter , also i m using Arducopter since 2013.

So , How to set Waypoints in RAM (volatile)  just like PX4 Firmware??????

#### Version
_What version was the issue encountered with_
ArduCopter 3.4.6

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

Copter

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Pixhawk

#### Logs
_Please provide a link to any relevant logs that show the issue_

magicrub, 2017-06-06:
By saying "in ram" you mean "clear on reboot", right?

RQRobert, 2017-06-06:
Yes exactly

magicrub, 2017-06-06:
How about a new param called "MIS_RETAIN" which defaults to 1 (enabled) which is the current behavior. When set to 0 it wipes the mission on boot.

RQRobert, 2017-06-06:
In addition to wiping out mission , also home location should clear too ?????

khancyr, 2017-06-06:
home position is reset on the arming position normaly !

RQRobert, 2017-06-06:
So the only thing left is a parameter like "MIS_RETAIN" , which solves the issue.

rmackay9, 2017-06-06:
Something similar came up in a discussion with @DonLakeFlyer.  Apparently QGC has a clear mission feature but it sometimes doesn't work with Copter because after an RTL, we sometimes consider that the mission is still running (I think).  So perhaps this could be a ground station feature and we just need to fix this bug that Don has raised?

RQRobert, 2017-06-06:
Randy

I was mainly interested in volatile nature of Waypoints (waypoints to be stored in RAM instead of FRAM).

magicrub, 2017-06-06:
They will never "be in RAM". We support huge mission lengths..

OXINARF, 2017-06-06:
There is already an issue requesting this, #5743, so I'm closing this one.

closed 2017-06-06 by OXINARF