ArduPilot/ardupilot #6390: LAND mode must not be changed to RTL by radio failsafe event

SergeyBokhantsev, 2017-06-06:
#### Issue details
Precondition:
Radio FS is configured to RTH.

Case:
Being far away from launch location pilot need to perform emergency landing.
It is impossible to do this in a reliable way: Pilot changes flight mode to LAND and when aircraft reaches low altitude then RC signal lost thus RC failsafe will be triggered (RTL).

Options:
1. Never change LAND mode on any FS events. Is there a reason exists to switch to RTH or to HOLD when current mode is LAND? 
2. Introduce new parameter so if set to TRUE then LAND mode should not be changed to RTL by Radio FS event.
2. Introduce new flight mode "Emergency land". Same as previous option this mode should not be interrupted by any FS.

#### Version
Master

#### Platform
[  ] Copter

Pedals2Paddles, 2017-06-06:
I'm not a fan of the example, because that's more like operator error.  But putting the example scenario aside, I think there may be some merit to option one for simplicity's sake.  If you're already landing, there is really no sense aborting that to go land somewhere else.

It could be as simple as landing on the opposite side of your back yard.  Then say the battery gets low and that RTL failsafe kicks in. That would be a waste of even more battery when you're already landing to begin with.

rmackay9, 2017-06-06:
Perhaps the answer though is to have the battery failsafe kick-in at the point that we need to start home in order to make it home with the remaining battery we have.  That would ensure that you always have the battery power reserved to RTL.

Re the opposite side of the back-yard, people probably already know but just in case, we have the rtl-cone-slope stuff at the moment so the vehicle wouldn't likely climb to the full rtl-alt.

SergeyBokhantsev, 2017-06-07:
@rmackay9 do you mean a special battery failsafe that will always do RTL if we are in the point-of-no-return? It would be an excellent improvement but it is not directly related to my issue.

I believe if operator decided to land here - it just must have a way to land here. For whatever reason. It may or may not be caused by battery voltage.

Pedals2Paddles, 2017-06-07:
Yup. The most recent (which is now quite old) solo branch of ArduCopter actually has this.  It will us altitude, RTL altitude, distance, RTL speed, up speed, down speed, and land speed to calculate how much power is needed to return home. And it will use that rather than only the arbitrary MAH or voltage. It's really neat, and I like it a lot. But it also needs some work to go mainstream. It's kind of a ham fisted method at the moment.  But it's something I have on my to-do list for the future.

geofrancis, 2017-06-07:
Sergey has a point, there is currently no way of landing a vehicle remotely if you have rc failsafe unless your in auto with failsafe set to continue mission.

Pedals2Paddles, 2017-06-07:
IMO a new mode isn't needed and would be very cumbersome. I think if the vehicle is in LAND mode, it should ignore radio and battery fail-safes. Plain and simple. It's already landing at a presumably safe location since the operator had already switched to land mode. No reason to abort that perfectly safe and swift landing to go land somewhere else. I'm actually surprised nobody thought of this until now. I believe it would be a very simple patch too. I might even take a stand at this evening depending on people's opinions here.

The two stage battery failsafe I'm not sure is a worthwhile complication. That would be a pretty significant change to a currently straight forward and predictable failsafe behavior. Complicating a failsafe with multiple actions may cause more wrecks than it prevents. Personally I believe this is just operator responsibility.

SergeyBokhantsev, 2017-06-07:
Thanks, 
actually I'm noticed this behavior on ArduCopter 3.2.1 (I'm fly APM board). So I decided to fix that in 3.2.1. It is really one-line fix there.
Then I just checked dev and saw that nothing changed. So I believe it could be a useful improvement for everyone.

ncx94, 2017-06-07:
I also think that when land have been triggered by the operator (or maybe in the end of an auto mission, same thing here), no failsafe should be able to kick in..

Pedals2Paddles, 2017-06-08:
I just took a stab at this here.  https://github.com/ArduPilot/ardupilot/pull/6406

SergeyBokhantsev, 2018-06-01:
Looks like that feature was lost in the chain of abandoned PRs and never merged.
Could we re-discuss on how it is possible to implement it? 
@Pedals2Paddles @rmackay9

auturgy, 2019-03-04:
https://discuss.ardupilot.org/t/gcs-failsafe-preventing-emergency-copter-landing/39247/2

Pedals2Paddles, 2020-01-07:
As of https://github.com/ArduPilot/ardupilot/pull/7432, there is now an option to continue landing on any failsafe.  Closed.

closed 2020-01-07 by Pedals2Paddles