ArduPilot/ardupilot #6391: Copter: Stick Mixing in guided mode.

ACPUK, 2017-06-07:
It would be a good as safety option to allow ‘Stick mixing’ during guided mode. As whilst operating a commercial operation there is always a pilot on the sticks in ALL modes. I appreciate it is a easy to switch mode from the RC but the natural response for a pilot it to operate the sticks to avoid an obstacle.

geofrancis, 2017-06-08:
I was just coming to post the same thing, my primary issue is not being able to use the throttle in guided to gain altitude if it starts dropping.

roque-canales, 2017-09-27:
+1 Yes we absolutely need to add this safety feature for guided mode.

ChrisBird, 2017-11-09:
I agree, I'll take a crack at this one if no one has started on it

rmackay9, 2017-11-09:
@tridge has done something for the skyrocket drones in RTL mode.  the skyrocket repo is here:https://github.com/SkyRocketToys/ardupilot.  We plan to merge all that into master eventually but it needs some review first.