ArduPilot/ardupilot #6393: EKF2 not disable more noisy IMU2 on pixracer, causing altitude rise.

hyperion11, 2017-06-07:
#### Issue details
EKF2 not disable more noisy IMU2 on pixracer, causing altitude rise.
As i think if IMU1 less noisy EKF2 must work with it then work with more noisy IMU2, isn't?
Or EKF2 is discounted, and i need try EKF3?

#### Version
3.5-rc7

#### Platform
[  ] All
[  ] AntennaTracker
[x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quad

#### Hardware type
Pixracer

#### Logs
http://discuss.ardupilot.org/uploads/default/original/2X/0/0fd5fe8d52481da65d994b2a2b1d9e804f3d122c.bin
[image]

OXINARF, 2017-06-07:
I'm not sure what you are reporting here. EKF doesn't disable IMUs, it chooses the best to work with.

It looks like you've posted in the forum, where this needs to be handled, this issues list is for feature requests and confirmed bugs.

closed 2017-06-07 by OXINARF

hyperion11, 2017-06-07:
So is it normal that EKF chooses an IMU with higher vibrations instead of selecting IMU with lower ones?

OXINARF, 2017-06-07:
Where is it choosing to use the second IMU?

hyperion11, 2017-06-08:
i see bad vibration level on imu2, and acceptable vibration level on imu1. But copter rise up. I think this is because ekf chooses a more noisy accelerometer. Or I'm wrong? 
Where in the logs we can see what data of which accelerometer at a particular time is used for INS?

OXINARF, 2017-06-08:
Look at the NKF4.PI field. Again, this is appropriate in the forum, not here.

hyperion11, 2017-06-09:
thank you. Sorry for possible offtopic here, but i have blank NKF4.PI and NKF9.PI field
[image]

OXINARF, 2017-06-09:
It's not blank, it is zero, meaning it is using the first IMU, not the second... Now, continue in the forum please.