ArduPilot/ardupilot #6394: EKF2 going crazy

njing, 2017-06-07:
#### Issue details
_Please describe the problem, or desired feature_
The alt changes by 30m in a few seconds. It'd reset and then go through it again. Please see the log below. It happens a couple of times.
#### Version
_What version was the issue encountered with_v3.5

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
quad X
#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
pixracer-like
#### Logs
_Please provide a link to any relevant logs that show the issue_
[12 12-31-1999 4-15-44 PM.bin.zip](https://github.com/ArduPilot/ardupilot/files/1057236/12.12-31-1999.4-15-44.PM.bin.zip)

rmackay9, 2017-06-07:
Normally we ask that beta testing reports go into the beta testing thread (http://discuss.ardupilot.org/c/arducopter/copter-3-5) so I'm going to close this issue if that's ok.  Sorry, but if we're not strict, our to-do list becomes a support forum.

I've had a quick look at the logs and as you say, the altitude climbs to about 25m and then resets back to zero and this repeats over and over.  I don't see anything wrong with the barometer or accelerometers.

I suspect the issue is that a number of EKF parameters are different from their defaults so I think these should be restored to their original values:

- EK2_ABIAS_P_NSE is it's 0.001, default is 0.005
- EK2_ACC_P_NSE is 0.5, default is 0.6
- EK2_ALT_M_NSE is 1, default is 3
- EK2_FLOW_M_NSE is 0.07, default is 0.25
- EK2_GYRO_P_NSE is 0.02, default is 0.03
- EK2_MAX_FLOW is 7, default is 2.5
- EK2_POSNE_M_NSE is 0.2, default is 1
- EK2_RNG_M_NSE is 0.1, default is 0.5
- EK2_VELD_M_NSE is 1, default is 0.7

Also I see that you're on an older release candidate (-rc4) but the lastest is -rc7.  Upgrading would be good because a lot of fixes went into -rc5.

Thanks for the report (but next time in the forum please!)

closed 2017-06-07 by rmackay9

njing, 2017-06-07:
Hi, Randy,

Thanks for your quick response. I really appreciate for your great effort,
and for the dedicated effort of the whole ardupilot team.

I'm new to this site and did not realize it's a todo list. I'll try your
suggestions and let's you the result in the forum.

Thanks again!

Neal