ArduPilot/ardupilot #6397: BAD LOGGING prearm fail when logging to file and mavlink

Pedals2Paddles, 2017-06-07:
#### Issue details
When the `LOG_BACKEND_TYPE`=3 (file and mavlink), and `LOG_DISARMED`=1 (enabled), there is a "BAD LOGGING" prearm failure. However this seems to be a false error, because logging is working fine.

If you change the backend type to only to file or only to mavlink, there is no failure. Only when trying to do both. And only when disarmed.

This is not just on mine. Everyone running master with this configuration is having the problem for months. I've worked around it by disabling the logging prearm check. But that's not a good fix.

Suggestions on what may be causing this?

#### Version
3.5+

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
N/A

#### Hardware type
Pixhawk 2.1 and IMX companion computer

njoubert, 2017-06-09:
We see the same thing. @ebethon

peterbarker, 2017-06-09:
The problem comes about because the mavlink backend is always "started", so the checks here: https://github.com/ardupilot/ardupilot/blob/master/ArduCopter/system.cpp#L498 mean we never call "start()", which means the file backend reports failure.  The arming checks call `start_logging` without making the same checks, so it *appears* to work - 'though I'm pretty sure you don't actually get a file log until you arm, contrary to the parameter!

I'm currently looking at cleaning all of this stuff up - but to do so in such a way that it could go into 3.5

Pedals2Paddles, 2017-06-09:
Cool. Thanks for working on this Peter!

peterbarker, 2017-06-09:
Well, there's now a PR for this issue.  Testing would be appreciated.

PR #6409 (2017-06-11): Correct health status of logging with LOG_DISARMED/multiple backends
Closes #6397
closed 2017-06-11 by OXINARF