ArduPilot/ardupilot #6400: UAVCAN GPS doesn't work anymore

akolobov, 2017-06-08:
#### Issue details
UAVCAN GPS (Zubax V2) doesn't work anymore on ArduPlane-- no 3D fix, bad GPS health. It used to work in February and earlier, so I don't know whether the issue is that the way it should be configured has changed since then (there are no changes in the documentation to indicate this) or that a software bug has been introduced.

#### Version
It has been broken since at least ArduPlane 3.8 beta5 (i.e., early May), possibly earlier. It is still broken as of June, 7, 2017.

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X ] Plane
[  ] Rover

#### Airframe type
Airplane, glider

#### Hardware type
Pixhawk

#### Logs

EShamaev, 2017-06-08:
Please advice on full setting's values for Zubax and also for AP. I will be glad to help as it is working. There might be some very little problem with settings.

akolobov, 2017-06-09:
Alright, so there a manual configuration fix for this issue, but I'd say that the bug is still there.

The heart of the matter, as I was explained on one of the forums, is that the new CAN stack for APM doesn't yet have the capability to assign node IDs automatically, so they need to be assigned manually for each CAN device. To do it for the Zubax unit:

- Connect it to your computer via USB and use Putty or something like that to access the Zubax unit's internal shell (see https://docs.zubax.com/usb and the Command-Line Interface section of this page: https://docs.zubax.com/zubax_gnss_2 for more detailed instructions).

- From the command line, run 

cfg set uavcan.node_id <any_number_between_2_and_125>
cfg save

Voila! Power cycle/reconnect the Zubax unit to your Pixhawk, and the Pixhawk will detect it.

Now, this is all dandy, but a) the fact that the above steps are *necessary* isn't documented anywhere and b) the CAN stack's currently inability to assign node IDs dynamically is broken functionality -- this used to work before.

So, despite the above workaround, it would be good to fix this bug properly. Please don't close this issue until a version of CAN with dynamic node ID assignment is checked in.

Thanks in advance to whoever will provide the fix!

EShamaev, 2017-06-09:
It will be done soon, however I consider it to be not a bug fix but extension of functionality.
More to that, in my opinion, dynamic node allocation is not that good feature.
But anyway, will be done.

akolobov, 2017-06-09:
Thanks Eugene! You understand this stuff way better than I, so out of curiosity: why is this not a good feature, and what was the motivation behind releasing it and then removing it?

EShamaev, 2017-06-09:
It was released in old PX4 firmware that was completely replaced. So it was not an act of removal.
As for good/bad, it is my personal feeling also backed up by lot's of standards and practices that in aviation everything should be strictly defined and rock solid. Auto node ID is nice but I there are some vulnerabilities if temporary malfunction of power supply to remote device happens (repowered in flight) and also with some temporary malfunctions of CAN bus wiring.

Evan1010, 2017-08-16:
Is there an updated procedure for setting up the Zubax GNSS 2 for Arduplane 3.8.0?

The existing (high level) settings for UAVCAN in Mission Planner 1.3.49 seem to be broken (unable to toggle enable UAVCAN under Initial Setup >> Optional Hardware >> UAVCAN), and I don't really understand the usage of the CAN_D*_PROTOCOL and CAN_P*_DRIVER parameters: 
[image]

I have set GPS_TYPE to 9, for UAVCAN.

I am working with the Pixhawk 2.1, using CAN2 (according to reports of CAN1 being mislabelled)

I attempted to use the serial console of the Zubax unit to assign a node ID, but the terminal spews out magnetometer, GPS, and barometer readings, and I can't seem to send the commands properly. Appropriate CLI behavior isn't particularly well documented on Zubax's knowledge base - essentially, it just says to connect and send commands: https://kb.zubax.com/display/MAINKB/NMEA+output+over+UART

Beyond that, how do I implement the onboard barometer (is it implemented in the 3D GPS code from the Zubax?) and magnetometer? Does the magnetometer automatically show up under configurable compasses, labelled external? Can I distinguish between two external compasses?

There's a lot of questions there, so if I need to summarize or rework, please tell me to do so.
I'd be glad to message anyone with a similar configuration (Zubax + PH2.1 + AP3.8.0 + MP1.3.49) to talk about proper configuration.

auturgy, 2017-10-15:
Can this be closed?

OXINARF, 2017-10-18:
I'm closing this as the OP was able to make it work as was Evan.

Having auto node ID in UAVCAN is something we should still pursue and I hope @EShamaev finds time to do it soon.

closed 2017-10-18 by OXINARF