ArduPilot/ardupilot #6401: Problem compiling firmware ArduPlane 3.6.0 using PX4 Toolchain

PedroHRPBS, 2017-06-08:
#### Issue details
Hello,
I'm trying to compile the ArduPlane 3.6.0 firmware for further testing with some modifications, but I keep getting these errors:

pedro_000@PedroHRPBS /c/Users/pedro_000/Documents/GitHub/ardupilot-ArduPlane-3.6.0/ArduPlane                            $ make px4-v2                                                                                                           fatal: ambiguous argument 'HEAD': unknown revision or path not in the working tree.                                     Use '--' to separate paths from revisions                                                                               fatal: ambiguous argument 'HEAD': unknown revision or path not in the working tree.                                     Use '--' to separate paths from revisions                                                                               fatal: ambiguous argument 'HEAD': unknown revision or path not in the working tree.                                     Use '--' to separate paths from revisions                                                                               // BUILDROOT=/c/Users/pedro_000/Documents/GitHub/ardupilot-ArduPlane-3.6.0/Build.ArduPlane HAL_BOARD=HAL_BOARD_PX4 HAL_BOARD_SUBTYPE= TOOLCHAIN=NATIVE EXTRAFLAGS=-DGIT_VERSION="HEAD" -I/c/Users/pedro_000/Documents/GitHub/ardupilot-ArduPlane-3.6.0/libraries/AP_Common/missing -DMAVLINK_PROTOCOL_VERSION=2 -DNUTTX_GIT_VERSION="HEAD" -DPX4_GIT_VERSION="HEAD" -DUAVCAN=1 -DHAVE_STD_NULLPTR_T=0 -I/c/Users/pedro_000/Documents/GitHub/ardupilot-ArduPlane-3.6.0/Build.ArduPlane/libraries/GCS_MAVLink/include/mavlink                                                                                             Checking modules                                                                                                        modules/gbenchmark/.git missing - need module init                                                                      + git submodule init                                                                                                    + for m in '$MODULE_LIST'                                                                                               + '[' -f modules/gbenchmark/.gitmodules ']'                                                                             + for m in '$MODULE_LIST'                                                                                               + '[' -f modules/gtest/.gitmodules ']'                                                                                  + for m in '$MODULE_LIST'                                                                                               + '[' -f modules/mavlink/.gitmodules ']'                                                                                + cd modules/mavlink                                                                                                    + git submodule init                                                                                                    You need to run this command from the toplevel of the working tree.                                                     + echo 'init of mavlink failed'                                                                                         init of mavlink failed                                                                                                  + git submodule status --recursive                                                                                      + exit 1                                                                                                                make: *** [CHECK_MODULES] Error 1                                                                                                                                 

I followed this tutorial: http://ardupilot.org/dev/docs/building-px4-with-make.html
I managed to compile the master with no problems. But as I'm using firmware 3.6 on my plane, I'd like to try some modifications to it.

Thanks in advance

magicrub, 2017-06-08:
Looks like a bit error, not necessarily anything wrong with the code. What commands did you use to get that branch and/or tag?

PedroHRPBS, 2017-06-08:
I downloaded it directly from Download ZIP, because I couldn't Clone the repository using Github for Desktop as I did with the master branch. 
Also, when I downloaded the files, all the folders inside the modules/ directory were empty. So I looked for the files on github, got the ZIP and pasted them where they were supposed to be.
I'm new to this, so I'm sure that I'm doing something wrong haha

OXINARF, 2017-06-08:
You need to use a git repository, downloading a zip doesn't work. If you cloned master, you can certainly checkout the 3.6 tag to a branch locally.

This issues list is for feature requests and confirmed bugs. For development questions please post in the forum (http://discuss.ardupilot.org) or ask in Gitter chat (http://gitter.im/ArduPilot/ardupilot).

closed 2017-06-08 by OXINARF