ArduPilot/ardupilot #6404: DataFlash/EKF2: getRangeBeaconDebug floating point exception when BCN_TYPE = 10 in SITL

rmackay9, 2017-06-08:
SITL is failing with a floating point exception if BCN_TYPE = 10 (SITL Beacon)

The exception happens in ../libraries/DataFlash/LogFile.cpp at line 1426 as it tries to multiply the "testRatio" by 100 and log it. https://github.com/ArduPilot/ardupilot/blob/master/libraries/DataFlash/LogFile.cpp#L1426

testRatio : (uint16_t)(100*testRatio)

This is a list of some values for the testRatio found during debugging.
ratio=0.000000
ratio=0.000000
ratio=0.000000
ratio=8520701837312.000000
ratio=8518961725440.000000
ratio=8602764443648.000000

The issue can be resolved in a hack-ish way by adding a constrain_float(testRatio,0,650) but there's probably a better solution.

peterbarker, 2017-06-08:
```
Breakpoint 1, NavEKF2_core::FuseRngBcn (this=0x8574e0)
    at ../../libraries/AP_NavEKF2/AP_NavEKF2_RngBcnFusion.cpp:166
166             rngBcnTestRatio = sq(innovRngBcn) / (sq(MAX(0.01f * (float)frontend->_rngBcnInnovGate, 1.0f)) * varInnovRngBcn);
(gdb) print varInnovRngBcn
$1 = 1.36711073
(gdb) print frontend->_rngBcnInnovGate
$2 = {<AP_Param> = {static _group_level_shift = 6 '\006', static _group_bits = 18 '\022', static _sentinal_key = 511, static _sentinal_type = 31 '\037', static _sentinal_group = 255 '\377', static _frame_type_flags = 1, static _storage = {type = StorageManager::StorageParam,
      total_size = 3840}, static _num_vars = 122, static _parameter_count = 852, static _var_info = 0x5a5630 <Copter::var_info>, static param_overrides = 0x857010, static num_param_overrides = 76, static k_EEPROM_magic0 = 80 'P', static k_EEPROM_magic1 = 65 'A',
    static k_EEPROM_revision = 6 '\006', static _hide_disabled_groups = true}, static vtype = AP_PARAM_INT16, _value = 500}
(gdb) print frontend->_rngBcnInnovGate.get()
$3 = (const short &) @0x7fdc4c: 500
(gdb) print innovRngBcn
$4 = -17065128
(gdb) print rngBcnTestRatio
$5 = 0
(gdb) n
(gdb) print rngBcnTestRatio
$6 = 8.52070184e+12
(gdb) print delta.length()
$7 = 16.7620373
(gdb) print rngBcnDataDelayed.rng
$8 = 17065144
(gdb) 
(gdb) print rngBcnDataDelayed 
$9 = {rng = 17065144, beacon_posNED = {x = 7.49175835, y = 14.9946489, z = 0}, rngErr = 1, beacon_ID = 0 '\000', time_ms = 39870}
(gdb) 

```

priseborough, 2017-06-08:
Can you send me a link to the log file that was generated by this test?

peterbarker, 2017-06-08:
[00000003.BIN.zip](https://github.com/ArduPilot/ardupilot/files/1062650/00000003.BIN.zip)

peterbarker, 2018-02-17:
I'm now unable to replicate this after reverting https://github.com/ArduPilot/ardupilot/pull/6408/commits/928534fcefb1bb7bcf58ffb61185f79d72af8967

Suggest we revert that commit and see if it breaks for anybody else now!

peterbarker, 2021-07-30:
This is ancient, all the logging has changed, EK3 is now our default, and we've autotested beacons for some time now.

We'll recreate this issue if it crops up again.

closed 2021-07-30 by peterbarker