ArduPilot/ardupilot #6405: Mission authorization

magicrub, 2017-06-08:
Allow Ardupilot to request flight authorization before arm. See pending PR https://github.com/PX4/Firmware/pull/7377.

On the first try to arm the vehicle will send a topic/MAVLink
message asking authorization the authorizer can be running anywhere
(compute board or ArduPilot it self) and it is responsible to request the
mission list and any other information to vehicle before send a
authorization response. If no response if given in 1 minute the
process is cancel.
If a ArduPilot got a VEHICLE_RESULT_ACCEPTED response it will allow
the pilot to arm the vehicle within 60 seconds.

Government entities like NASA are working in systems to managed the aero space for drones(https://utm.arc.nasa.gov/) but this also can have another uses by private companies.

Here we have a prof of concept application that authorizes mission based on the altitude of the last mission item. 01org/mavlink-router#94

peterbarker, 2017-06-13:
This was discussed on the dev call today.

It was generally agreed that a good way forward is to use a challenge/response protocol based on mavlink2 and its signing key.

So the vehicle generates and remembers a nonce (32-bit?).  It sends out an authorized-to-fly request containing the nonce.

Anybody who feels authorized to authorize the vehicle to fly (be it a companion computer talking to e.g. Altitude Angel or a GCS doing the same) can respond to the request by hashing the nonce with the mavlink2 signing key and sending it in a flight-is-authorized packet.

The drone listens for flight-is-authorized packets and checks to see if the remembered nonce and the signing key hash to the value in the packet, passing the prearm check if they match.

Discuss.

peterbarker, 2021-07-30:
We support auxillary authorization arming now in AP_Arming.

Any more-specific protocol can be opened as a new PR.

closed 2021-07-30 by peterbarker

rmackay9, 2021-07-30:
We definitely still need to have the mavlink based arming authorisation.  Hopefully that enhancement request in still in some other issue.